SmoothWAN/SmoothWAN #1: Generic x86_64 build?

CroutonCrusader, 2021-12-27:
Hi -- I happened across your Reddit post and hate that you were banned, because this project looks very promising and helpful. I'm unsure how else to contact you, so I'm opening this up as an "issue".

I'm currently running Ubuntu Desktop with Speedify client installed on a Qotom-Q555G6-S05 x86 mini PC.

I'd like to try out the x86 build. Will you be posting an x86 build?

TalalMash, 2021-12-27:
Hi! 
Done, I'd appreciate a small feedback if it works well with the Qotom, interesting hardware.

CroutonCrusader, 2021-12-27:
I loaded smoothwan-0.92-x86-64-generic-ext4-combined-efi.img to a USB drive and booted the Qotom. This is where it stops during the boot process
[image]
.

TalalMash, 2021-12-27:
If pressing enter gets you to the console then its working properly, the Web UI is now accessible at 192.168.3.1 from one of the ethernet ports depending on hardware.

CroutonCrusader, 2021-12-27:
I pressed enter and it gave me a console. I have eth0 connected to my T-Mobile gateway, eth4 connected to my switch, and eth5 connected to a Peplink router. My Peplink router is reporting "No cable connected" and when I type "ip a", it lists all ports as "DOWN"
[image]
.

TalalMash, 2021-12-27:
Hmm, root@(none) suggests that there is mounting issue, can you use the non EFI image and switch to BIOS? I just tested EFI with 2 UEFI systems and it worked well, maybe the image is corrupted. (?)