SmoothWAN/SmoothWAN #16: Port forwarding

umekzg, 2022-03-20:
Hi!

Neat piece of software!

As i have a dedicated server i also have option for port forwarding.

I'm aware that it would be double NAT but is it possible to fwd a port to LAN device from speedify in current build?

Also are USB 4G modems supported?

Thx!

umekzg, 2022-03-20:
scrap this... i didn't see discussions.....

TalalMash, 2022-03-20:
I've added VPN port forward quick configuration page a while ago, its not in the Wiki yet as I don't have a dedicated plan to test, please let me know if it works
[image]
[image]

With regarding USB modems, I'll update README and replace the word "cellular" with "4G" as it seems to be confusing. 
RNDIS/USB-ethernet-like modems are supported by Plug and Play function and the Multi-WAN configration pages (including renaming) 
QMI and NCM can be set up manually in "Interfaces" page only as they're uncommon and require specific settings from ISP.

umekzg, 2022-03-20:
Wow nice!
Testing tomorrow!

umekzg, 2022-03-21:
Can you make a bcm2710 build? I don't have any rpi4 laying around.

TalalMash, 2022-03-22:
> Can you make a bcm2710 build? I don't have any rpi4 laying around.

I don't have any older rpi as they're very slow, the speed matters for latency and bufferbloat, not just bandwidth. I'll build one soon for experimenting but without multi WAN USB renaming UI. Plug and play is still supported.

TalalMash, 2022-03-23:
> Can you make a bcm2710 build? I don't have any rpi4 laying around.
Added in latest release. Wi-Fi may need additional configuration, use ethernet.
Set interfaces manually in "Interfaces" page, the Multi-WAN USB app is not functional as I don't have the hardware to test.

closed 2022-03-27 by TalalMash

TalalMash, 2022-04-07:
> Can you make a bcm2710 build? I don't have any rpi4 laying around.

Any updates regarding the RPi3 build?

umekzg, 2022-04-07:
Actually i was in covid for 10 days...
I'm now catching up on everything...

When i do test i will post results here.

TalalMash, 2022-04-07:
> Actually i was in covid for 10 days... I'm now catching up on everything...
> 
> When i do test i will post results here.

Wow thats unfortunate, hope for a speedy recovery!

umekzg, 2022-06-05:
I managed to source rpi4 and tried the port fwd. Didn't work.

TalalMash, 2022-06-05:
Thank you for reporting.