SmoothWAN/SmoothWAN #20: Netgear X6 R8000

nmz787, 2022-04-11:
An R8000 image is listed here, present too is the imagebuilder item:
https://downloads.openwrt.org/releases/21.02.1/targets/bcm53xx/generic/

but I then saw there's no R8000 in the smoothWAN repo's `devconfigs` folder. What do I do now?

I've already got an R8000 that has been pretty reliable with freshtomato, and which I've had open-wrt running on before (the image linked above). They seem well powered, and cost reasonable in used markets. They've got multiple USB ports on them.

TalalMash, 2022-04-11:
Unfortunately the resources on these are very scarce, and no AES acceleration to back it up.
NanoPi R4S is supported in the next release, should cost less, and much more powerful than a Pi 4, the reason for support is due to endless chip shortage with the Pi's, although Pi's are primarily supported due to global availability, not for performance or cost.
Note that OSS for wireless modules are majorly not optimized/reliable, stock firmware is recommended for routers/AP when used with SmoothWAN.

nmz787, 2022-04-12:
Would I still be able to achieve around 30-50 MBPS max? I am not looking for throughput as much as for bonding multiple LTE modems for better latency (and any bandwidth aggregation/increase would be welcome).