SmoothWAN/SmoothWAN #22: Change LAN IP

neosaadmin, 2022-04-12:
Changing LAN IP results in the old IP 172.17.17.2 still being active and showing speedify not able to load on the overview when viewing at say 192.168.3.1, but speedify is running and active at 172.17.17.2 still.

TalalMash, 2022-04-12:
Hello, good issue, changing IP requires a post changing script that is being worked on, the current workaround is using a unique IP such as the default. Few components should be run as services for clean restarts (e.g firewall rules, tailscale, LXC containers such as Pi-hole).
Will be added in 0.99.6

TalalMash, 2022-04-16:
> Changing LAN IP results in the old IP 172.17.17.2 still being active and showing speedify not able to load on the overview when viewing at say 192.168.3.1, but speedify is running and active at 172.17.17.2 still.

In the meantime you can simply change the IP in Firewall -> Custom Rules

TalalMash, 2022-04-25:
Temporary till next version:
In Web UI Terminal app, login and type:
`wget https://raw.githubusercontent.com/TalalMash/SmoothWAN-web/main/switchto192.sh`
`cat switch192.sh` (check content)
`sh switch192.sh` 
To revert replace 192 with 172 in URL.

TalalMash, 2022-08-06:
Fixed in latest release, reboot after changing IP.
Re-open if there are issues.
Thanks.

closed 2022-08-06 by TalalMash