SmoothWAN/SmoothWAN #34: Using SSD storage

TalalMash, 2022-05-05:
### Discussed in https://github.com/TalalMash/SmoothWAN/discussions/32

<div type='discussions-op-text'>

<sup>Originally posted by **hle5128** May  3, 2022</sup>
I sometimes get a 502 lousy gateway in Luci and need to reboot to bring back the admin.
Do you know the reasons?
[image]
</div>

TalalMash, 2022-05-05:
Hello, this sounds like sdcard issue or corruption, try reflashing or changing sdcard model/brand.

_Originally posted by @TalalMash in https://github.com/TalalMash/SmoothWAN/discussions/32#discussioncomment-2681381_

I'm using SSD card booting, reboot does let me back in though,
i dont have problem with this in the past with ras pi 4

_Originally posted by @hle5128 in https://github.com/TalalMash/SmoothWAN/discussions/32#discussioncomment-2681542_

> I'm using SSD card booting, reboot does let me back in though, i dont have problem with this in the past with ras pi 4

I see.. There is limited root disk detection in OpenWRT, and the additional filesystem check script targets sdcards. Maybe it's in read-only mode due to some error? Please post the last 30 lines or so of `dmesg` output here, make sure you open an SSH session before it errors again.

You have to modify the filesystem check-repair to work with SSD by editing in terminal the file /lib/preinit/80_mount_root and change `/dev/mmcblk0p2` to `/dev/sda2` at line 21.

_Originally posted by @TalalMash in https://github.com/TalalMash/SmoothWAN/discussions/32#discussioncomment-2681581_

yup, i have done that, will get the logs for you next time. thanks

_Originally posted by @hle5128 in https://github.com/TalalMash/SmoothWAN/discussions/32#discussioncomment-2690262_

TalalMash, 2022-05-05:
_Originally posted by @hle5128 in https://github.com/TalalMash/SmoothWAN/discussions/33_

After setting up vpn bypass, refresh and the field is empty.
reboot still has same problem

[image]

TalalMash, 2022-05-05:
Can you do `cat /tmp/fsckok` and post the results, if the system is still online, the last 30 lines of `dmesg` is also useful, this still seems like a storage issue, maybe the previous corruption prior to the fix caused some issues. Fresh install config /etc/config/vpnbypass:
```
config vpnbypass 'config'
        option enabled          '0'
```

hle5128, 2022-05-05:
I reflash the firmware solved the problem.
Also, is there a way we have the speedify build on to gl inet router?

TalalMash, 2022-05-05:
> I reflash the firmware solved the problem. Also, is there a way we have the speedify build on to gl inet router?
> […](#)

Good to hear! 
The [Brume-W](https://www.gl-inet.com/products/gl-mv1000/) seems suitable albeit having 3 ethernet ports only, 
PC engines [APU4D4](https://www.pcengines.ch/apu4d4.htm) is much more powerful and around the same price, it seems to be the cheapest allrounder hardware other than the Qotoms, supporting Wi-Fi NICs and 4G modems (dual active SIM) without configuration.
Feel free to reopen if the issue persists.

closed 2022-05-05 by TalalMash

hle5128, 2022-05-06:
> The Brume-W seems suitable albeit having 3 ethernet ports only, the CPU is a bit weak but AES is supported so it may get near a Pi 4, I'd need to test the hardware for the build or someone test it.
> Edit: Regarding APU2 scratch that, different hardware on new versions due to shortage.
> And Brume-W has reached EOL, I'm willing to try Flint

Seem like flint is good for combination for router, wifi and speed. Im planning to get one, the cost is cheaper than pi4, its around $90 vs pi 4 $120 plus other usb lan etc...
Im buying flint now arrive tomorrow. Im open to test.
Let me know.

hle5128, 2022-05-06:
I'm waiting for the slate AX to come out to purchase and test, meanwhile I have the flint to test tomorrow. 
Let me know what you need. I have 2x 5G modems and 1 Cable to test the speed as well. My cable is 500~Mbps and 2x 5G around 180Mbps each, when testing pi4 sppedify, i only got half of that speed, wonder if its hardware limits or not, (max around 150ish Mbps down for combining 2x 5G

TalalMash, 2022-05-06:
> I'm waiting for the slate AX to come out to purchase and test, meanwhile I have the flint to test tomorrow. Let me know what you need. I have 2x 5G modems and 1 Cable to test the speed as well. My [...]

Interesting results, you may have to disable encryption and use the built-in speed test to see if the Pi is limiting, Speedify takes few seconds to ramp up the speed due to weighing algorithms. You can also "rate limit" one of the connections slightly lower than the other, having two connections at the same exact speed if both set as primary can slow down the ramp-up time.
I'm porting Gi Net firmware for PC to build an official package, will provide the app ".ipk" file here, it's installable via the WebUI, no need to use SmoothWAN firmware for this if it works well.

hle5128, 2022-05-06:
Cant wait,
Let me test on pi4 base on ur suggestion first and will report it back, i also have brume-w here and the flint arrives tomorrow, so i can test both comparing to pi4b and give you more result and details information. I also have nano pi rs4 arrives tomorrow to test as well. 
I would guess if its just the ipk, we just need to configure macvlan (brume and flint)

TalalMash, 2022-05-06:
Sounds great! Instead of ipk I'll make an automated script that enables LuCI (GI net WebUI intact) and copy over SmoothWAN files, the firmware is proprietary hence why flashable images aren't possible yet but IPQ4/8xx SoC official OpenWRT supports is coming in the few months.
The Nano Pi R4S is much more powerful than the Pi and Gi, but the Gi seems the only option with decent integrated Wi-Fi if needed. Haven't released R4S image yet, will do with the Gi net script.

hle5128, 2022-05-06:
Oh I thought there is an image rs4 already available, maybe mistake that i Read the docs saying “compatible” only, ah just check only ras pi 4 and x86.
Ok keep us updated. 
Thanks

TalalMash, 2022-05-07:
@hle5128 My bad, I've updated the pre-release images and added an issue page, it's hidden, you need to click on "Releases", the pre-release tag is present.

hle5128, 2022-05-07:
D you know every now then the speedify turn off itself, almost every day in morning i need to login and turn on again. I have the function disconnect on exit off but still.

TalalMash, 2022-05-07:
> D you know every now then the speedify turn off itself, almost every day in morning i need to login and turn on again. I have the function disconnect on exit off but still.
> […](#)

Maybe the connect at startup has been accidentally switched off? And the Pi is restarting due to power issue. 
To generate Speedify logs, do `tar -zcvf /www/spdlog.tar.gz /usr/share/speedify/logs/` then visit http://172.17.17.2/spdlog.tar.gz 
Upload here to check for the reason if you want to check instead, the last status is usually in the latest speedify_bydate file.
Next version will have a button to automate this.