SmoothWAN/SmoothWAN #35: [Testing needed] Rockchip RK3399 SBC support

TalalMash, 2022-05-07:
Support for:
- NanoPi R4S
- Pine64 RockPro64
- Radxa RockPi

Features to implement:
- Simplified MACVLAN app due to limited ports e.g NanoPi
- Default Wi-Fi AP settings (need a tester)
  - Post your `/etc/config/wireless` on fresh install
- OLED display support and I2C (need a tester)
  - Post your `/sys/bus/i2c/devices/`
- Check if SpeedifyAES acceleration is working properly (need a tester)
  - Post your last 10 lines of `cat /usr/share/speedify/logs/speedify_* | grep hardware`
- Check if the root filesystem is correct for `fsck` on startup (need a tester)
  - Post your `mount` command output (for SD or/and EMMC)
- USB port numbering utility for identification is WIP for Radxa/Pine.
- Apply [hardware-optimization](https://github.com/armbian/build/blob/master/packages/bsp/common/usr/lib/armbian/armbian-hardware-optimization) on final image and test before release.
- Only OpenWRT version 22.x and higher supports RK3399, and with the new firewall rules, Speedify UI may not work properly and will require adjustments in `firewall.custom` file.

Testing images are uploaded as prerelease in Relases page
Sysupgrade and Factory image is the same on this platform, flash using balenaEtcher.

hle5128, 2022-05-09:
given an error disk after flash using balenaEtcher

hle5128, 2022-05-09:
https://share.getcloudapp.com/geumLqOm

hle5128, 2022-05-09:
the same for sdcad

TalalMash, 2022-05-10:
Thanks for the video, I couldn't see the error. Is it the Mac pop up? That's normal as it's a Linux filesystem, unreadable by Mac.

hle5128, 2022-05-10:
/dev/disk4 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *63.9 GB    disk4
   1:                      Linux ⁨⁩                        268.4 MB   disk4s1
   2:                      Linux ⁨⁩                        1.9 GB     disk4s2
                    (free space)                         61.6 GB    -

jimmyle-m16@Jimmy-MBP-M1M ~ % diskutil mount disk4s1
Volume on disk4s1 failed to mount
Perhaps the operation is not supported (kDAReturnUnsupported)
If you think the volume is supported but damaged, try the "readOnly" option
jimmyle-m16@Jimmy-MBP-M1M ~ %

TalalMash, 2022-05-10:
I'm not aware that Mac OS can mount EXT4, can you elaborate?
Here it is after flashing with balenaEtcher on Windows, displayed on Ubuntu:
[image]

hle5128, 2022-05-10:
alright, im in , i will post all information you need here in few mins

hle5128, 2022-05-10:
root@SmoothWAN:~# mount
/dev/mmcblk1p2 on / type ext4 (rw,noatime,errors=continue,commit=30)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,relatime,net_cls)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,relatime,rdma)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,name=systemd)

hle5128, 2022-05-10:
Wifi:

Fri Jan 18 09:02:26 2013 kern.info kernel: [  738.086282] usb 5-1: new high-speed USB device number 2 using xhci-hcd
Fri Jan 18 09:02:27 2013 kern.warn kernel: [  738.317366] RTL871X: rtw_ndev_init(wlan0)
Fri Jan 18 09:02:27 2013 daemon.debug ModemManager[12631]: hotplug: event reported: action=add, name=wlan0, subsystem=net
Fri Jan 18 09:02:27 2013 daemon.info ModemManager[12631]: hotplug: no need to wait for modem at sysfs path /sys/devices/platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb5/5-1
Fri Jan 18 09:02:30 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:02:30 2013 daemon.info [6316]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb5/5-1': not supported by any plugin
Fri Jan 18 09:02:46 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:03:02 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:03:18 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:53:58 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:54:14 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:54:30 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:54:46 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:55:02 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:55:18 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:55:34 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:55:50 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:56:06 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:56:22 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:56:38 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:56:54 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:57:10 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:57:26 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:57:42 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:57:58 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:58:09 2013 daemon.info uwsgi-luci: Fri Jan 18 08:58:09 2013 - received message 0 from emperor
Fri Jan 18 08:58:09 2013 daemon.info uwsgi-luci: SIGINT/SIGTERM received...killing workers...
Fri Jan 18 08:58:09 2013 daemon.err uwsgi[3022]: Fri Jan 18 08:58:09 2013 - [emperor] bringing back instance /etc/uwsgi/vassals/luci-webui.ini to on-demand mode
Fri Jan 18 08:58:09 2013 daemon.err uwsgi[3022]: Fri Jan 18 08:58:09 2013 - [emperor] curse the uwsgi instance /etc/uwsgi/vassals/luci-webui.ini (pid: 10793)
Fri Jan 18 08:58:11 2013 daemon.err uwsgi[3022]: [uwsgi-emperor] /etc/uwsgi/vassals/luci-webui.ini -> back to "on demand" mode, waiting for connections on socket "/var/run//luci-webui.socket" ...
Fri Jan 18 08:58:14 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:58:30 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:58:46 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:59:02 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:59:18 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:59:34 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 08:59:50 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:00:06 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:00:12 2013 daemon.err uwsgi[3022]: *** has_emperor mode detected (fd: 7) ***
Fri Jan 18 09:00:12 2013 daemon.err uwsgi[3022]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
Fri Jan 18 09:00:13 2013 daemon.err uwsgi[3022]: Fri Jan 18 09:00:13 2013 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Fri Jan 18 09:00:13 2013 daemon.err uwsgi[3022]: Fri Jan 18 09:00:13 2013 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Fri Jan 18 09:00:13 2013 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 12456)
Fri Jan 18 09:00:13 2013 daemon.err uwsgi[3022]: Fri Jan 18 09:00:13 2013 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Fri Jan 18 09:00:22 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Fri Jan 18 09:00:38 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
[... cut ...]
Fri Jan 18 09:03:18 2013 daemon.err odhcpd[5395]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
^C
root@SmoothWAN:~# cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb5/5-1/5-1:1.0'
	option band '5g'
	option htmode 'NOHT'
	option channel 'auto'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option key 'appletintin'

root@SmoothWAN:~#

TalalMash, 2022-05-10:
Thanks! Is the AP visible and working?

hle5128, 2022-05-10:
> Thanks! Is the AP visible and working?

i dont see AP yet, as soon i submit AP information, everything is dead on GUI, root still working, though i still dont see my AP yet, i wonder if i need a reboot, one second

TalalMash, 2022-05-10:
Did you use the WiFi basic page? Wi-Fi settings should be untouched for development, what do you mean by submit?

hle5128, 2022-05-10:
yes i went to wifi basic page to enter the AP name and password, so reboot does nothing and LAN won't give out any more IP nor i dont see AP on wifi. Tried manually IP lan still not let me in RS4, (no other modem plugged in) just LAN to my computer and Wifi USB

hle5128, 2022-05-11:
root@SmoothWAN:~# ls /sys/bus/i2c/devices/
0-001b  0-0040  0-0041  2-0051  i2c-0   i2c-1   i2c-2   i2c-7

hle5128, 2022-05-11:
[image]

there is WAN/WAN6 under eth0 on none firewall.
WAN has no interface
I have LAN plugged in to my laptop
WAN plugged to modem

hle5128, 2022-05-11:
after installed speedify i get this screen and error. restart does the samething

Wed May 11 02:28:15 2022 daemon.err nginx[3024]: 2022/05/11 02:28:15 [error] 3045#0: *64 open() "/www/spdui/WorkSans-Regular.ttf" failed (2: No such file or directory), client: 172.17.17.160, server: _lan, request: "GET /spdui/WorkSans-Regular.ttf HTTP/1.1", host: "172.17.17.2", referrer: "http://172.17.17.2/spdui/?uid=186371"
Wed May 11 02:28:15 2022 daemon.err nginx[3024]: 2022/05/11 02:28:15 [error] 3045#0: *78 open() "/www/spdui/WorkSans-SemiBold.ttf" failed (2: No such file or directory), client: 172.17.17.160, server: _lan, request: "GET /spdui/WorkSans-SemiBold.ttf HTTP/1.1", host: "172.17.17.2", referrer: "http://172.17.17.2/spdui/?uid=186371"
Wed May 11 02:28:15 2022 daemon.err nginx[3024]: 2022/05/11 02:28:15 [error] 3049#0: *79 open() "/www/spdui/WorkSans-Bold.ttf" failed (2: No such file or directory), client: 172.17.17.160, server: _lan, request: "GET /spdui/WorkSans-Bold.ttf HTTP/1.1", host: "172.17.17.2", referrer: "http://172.17.17.2/spdui/?uid=186371"
Wed May 11 02:28:15 2022 daemon.err nginx[3024]: 2022/05/11 02:28:15 [error] 3049#0: *80 open() "/www/spdui/WorkSans-Light.ttf" failed (2: No such file or directory), client: 172.17.17.160, server: _lan, request: "GET /spdui/WorkSans-Light.ttf HTTP/1.1", host: "172.17.17.2", referrer: "http://172.17.17.2/spdui/?uid=186371"
Wed May 11 02:28:20 2022 daemon.err odhcpd[5421]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
[image]

hle5128, 2022-05-11:
after reboot,  got 502 gateway page and can not login to root.
jimmyle-m16@Jimmy-MBP-M1M ~ % ssh root@172.17.17.2
kex_exchange_identification: read: Connection reset by peer
Connection reset by 172.17.17.2 port 22
jimmyle-m16@Jimmy-MBP-M1M ~ %

hle5128, 2022-05-11:
after reflash the firmware, i keep getting the self ip from lan port...no matter what i do, manually assign ip doesnt help,

TalalMash, 2022-05-11:
I'll build a new image with the required changes, are you using SD card?

TalalMash, 2022-05-11:
> Yes, Sd card, seem like no matter how many time i reflashed sd card, notthing work on given ip, only work 1st time after that i dont understand why it wont give out ip anymore even though its a new flash
> […](#)

Thanks for the testing, I'm considering getting one soon as its pretty difficult to diagnose remotely.
I added your result notes in the main post.
I'll send a squashFS image with the next build, I'll build v22 for x86 and test few things too.

hle5128, 2022-05-11:
> Simplified VLAN app due to limited ports e.g NanoPi
> Default Wi-Fi AP settings (need a tester for Radxa/Pine)
> Post your /etc/config/wireless on fresh install, skip for R4S.
> OLED display support and I2C (need a tester, done by @hle5128)
> Post your ls /sys/bus/i2c/devices/
> Check if the root filesystem is correct for fsck on startup (need a tester, done by @hle5128)
> Only OpenWRT version 22.x and higher supports RK3399, and with the new firewall rules, Speedify UI may not work properly and will require adjustments in firewall.custom file, will move from iptables to firewall4 if iptables-nft does not work properly. (need a tester, done by @hle5128)

Yes, what a bumper i couldnt test speedift, GUI wouldn't load up, then doesn't give up after that. Seem like r4s is way faster performance than ras pi 4.

TalalMash, 2022-05-11:
> Yes, what a bumper i couldnt test speedift, GUI wouldn't load up, then doesn't give up after that. Seem like r4s is way faster performance than ras pi 4.
> […](#)

SquashFS may work, Ill upload in few minutes.

TalalMash, 2022-05-11:
@hle5128 Uploaded a SquashFS image, fixed Speedify UI (tested on x86 v22) and default firewall for WAN.

hle5128, 2022-05-11:
do i just flash with balenEtcher as usual ? 
Downloading it now

TalalMash, 2022-05-11:
> do i just flash with balenEtcher as usual ? Downloading it now
> […](#)

Yes

hle5128, 2022-05-11:
keep getting self-assigned IP, (have nothing plugged in except the lan port to my computer)
know any ideas on how to access and get the logs out?

TalalMash, 2022-05-11:
Do you get the same thing when plugged to WAN? It's a bit odd that it used to work, check hardware/SD card.

hle5128, 2022-05-11:
yes samething and also usb lan
so i reflash with friendlywrt comes with is on same sd card, and i got the ip right away and the OS booted just fine.
something couldnt boot for sure, (it did first time that i got the logs for you and that was it)

TalalMash, 2022-05-11:
Ok i'll rebuild with friendlywrt UCI defaults, it should be close to stock

hle5128, 2022-05-11:
and the usb wifi Realtek RTL8812AU is working as well on friendlyopenwrt latest version

https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R4S#Work_with_USB_WiFi_Device

https://drive.google.com/drive/folders/1C0-QsuJRXYfRpcHEU6lCM_lV4PT1z6t-

hle5128, 2022-05-11:
> You said you used Wireless (Basic) page, try the regular Wireless page like the FriendlyWRT Wiki instructions.
> Anyways I'll upload the new build in a minute.

im just testing friendlywrt and the USB wifi is working, but i guess they updated the driver in friendwrt os vs openwrt
i will wait for your build to test fully with wifi and speedify and see some speed.

TalalMash, 2022-05-11:
https://forum.openwrt.org/t/rtl8812au-and-or-rtl8814au-drivers/67/185 It seems to require out of tree hack/patch and it's quite unstable, fast but lossy, friendlywrt is using an old kernel which may break few things.
The supported Wi-Fi hardware is in the Wiki.
You can try the new uploaded release.

hle5128, 2022-05-11:
alright, so plugged in lan port, has no light activity indication whatsoever, and no ip, no self-assigned, port just dead.
if plugged in wan, then we got light activity but no ip, throwing self-assigned ip, manually assigned ip to 172.17.17.100, does not route to router 172.17.17.2

TalalMash, 2022-05-11:
In friendlyWRT, what is the bridged root device for LAN? Can you try the official OpenWRT image too: https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/

hle5128, 2022-05-11:
here you go for friendlywrt
https://share.getcloudapp.com/yAuKd94O

hle5128, 2022-05-11:
i cant boot from [friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz]
but i can boot from [friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz]

 LAN ip now has light activity but no IP, just self-assigned IP.
WAN also has light activity but the same thing, no ip, and just self-assigned IP.

TalalMash, 2022-05-11:
So now friendlyWRT stopped working?

hle5128, 2022-05-11:
no,
friendlywrt works just fine, i mention about firmware you want me to try in:
https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/ <https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/>

TalalMash, 2022-05-11:
Ok I'll make a script that installs SmoothWAN apps on GL net and RK3399 tomorrow, you can try this lastly which has an update two days ago: https://chef.libremesh.org/?version=22.03-SNAPSHOT&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r4s

hle5128, 2022-05-11:
the build openwrt @ https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/
give a same problem with smoothwan firmware, no ip address on both lan/wan, only self assigned ip
sys/pwr are on, i wonder if its finish booting, otherwise why it wouldnt give out ip unless its stuck. Will try the snapshot you give now

TalalMash, 2022-05-11:
Do you have the 1GB or 4GB version? I think the official OpenWRT is for 4GB only :/

hle5128, 2022-05-11:
Mine is 4G version,

hle5128, 2022-05-11:
i tried the snapshot both files ex4 and squashfs wont boot, SYS light not on, tried power off/on many times on both files.

hle5128, 2022-05-11:
ok sseem like booting with LAN port plugged wont boot, i have to boot without LAN/WAN plugged in, then once i see the SYS is on, i can plug the LAN in, i got LAN light on now and connected on this build:

https://chef.libremesh.org/?version=22.03-SNAPSHOT&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r4s

hle5128, 2022-05-11:
but it gave private ip and ip6, i tried 192.168.2.1 and 192.168.1.1 manually wont connect to the router.
[image]

TalalMash, 2022-05-11:
> [image]
> 
> i tried the snapshot both files ex4 and squashfs wont boot, SYS light not on, tried power off/on many times on both files.

What is that photo?
Have you tried a different SD card brand? Non-mainline kernel seems to use different voltage for sdio and disable UHS mode.
Official OWRT without mainline: [OpenWrt-AO-NanoPiR4S-full-21.02-20220501-squashfs.zip](https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases/download/OpenWrtAO-21.02-20220501/OpenWrt-AO-NanoPiR4S-full-21.02-20220501-squashfs.zip)

hle5128, 2022-05-11:
> Uploaded wrong photo in my download folder.
> Yes i tried few sd cards, the one im trying is the one working with friendlywrt

I will try again when getting home. 
Keep you posted. Seem like r4s is very picky somehow on booting comparing to ras pi 4

hle5128, 2022-05-11:
none of these images are working, can we just install and build speedify on friendlywrt?

hle5128, 2022-05-12:
alright, im able to boot from your earlier image "smoothwan-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img"

LAN/WAN both lights are working, got the ip right away after 1st boot, but couldnt login GUI, 502 Bad Gate way.

but able to login and got you these errors:

Thu May 12 01:50:26 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:26 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:27 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:27 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:28 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:28 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:29 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:29 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:30 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:30 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:31 2022 daemon.info ModemManager[8543]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:31 2022 daemon.info ModemManager[8543]: hotplug: ModemManager not yet available
Thu May 12 01:50:31 2022 daemon.err ModemManager[8543]: hotplug: couldn't report initial kernel events: ModemManager not running
Thu May 12 01:50:32 2022 daemon.err odhcpd[5349]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Thu May 12 01:50:37 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:37 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available
Thu May 12 01:50:38 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:38 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available
Thu May 12 01:50:39 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:39 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available
Thu May 12 01:50:40 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:40 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available
Thu May 12 01:50:41 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:41 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available
Thu May 12 01:50:42 2022 daemon.info ModemManager[8801]: hotplug: checking if ModemManager is available...
Thu May 12 01:50:42 2022 daemon.info ModemManager[8801]: hotplug: ModemManager not yet available

hle5128, 2022-05-12:
root@SmoothWAN:~# mount
/dev/mmcblk1p2 on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/mmcblk1p2 on /overlay type f2fs (rw,noatime,lazytime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=reuse,fsync_mode=posix)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,relatime,net_cls)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,relatime,rdma)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,name=systemd)

root@SmoothWAN:~# cat /etc/config/network

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd00:0bbb:afd0::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1'

config device
	option name 'eth1'
	option macaddr '82:34:28:38:77:7e'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '172.17.17.2'

config device
	option name 'eth0'
	option macaddr '80:34:28:38:77:7e'

config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'

config interface 'wan6'
	option device 'eth0'
	option proto 'dhcpv6'

config interface 'Speedify'
	option proto 'none'
	option device 'connectify0'

root@SmoothWAN:~#

hle5128, 2022-05-12:
after reboot, it back to same issues again, both WAN/LAN give no ips, SYS/PWR on, LAN/WAN no light.
Reflash again, then it doesnt work 2nd time. 
Seem like its only work 1st time with new sd card? Im getting serial terminal tomorrow to see the debugs whats going on

hle5128, 2022-05-12:
look like it doesnt fully complete boot thats why it gets no IP,

hle5128, 2022-05-12:
this is information from friendlywrt i got.

root@FriendlyWrt:~# lsb_release -a || cat /usr/lib/os-release || cat /etc/os-release
-bash: lsb_release: command not found
NAME="OpenWrt"
VERSION="21.02.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.2"
VERSION_ID="21.02.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16495-bf0c965af0"
OPENWRT_BOARD="rockchip/armv8"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.2 r16495-bf0c965af0"

root@FriendlyWrt:~# uname -a
Linux FriendlyWrt 5.15.25 #1 SMP PREEMPT Tue Mar 1 16:28:57 CST 2022 aarch64 GNU/Linux

root@FriendlyWrt:~# ls /dev/mmcblk? | xargs -IDEV -n1 parted DEV print
Model: SD SR64G (sd/mmc)
Disk /dev/mmcblk1: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name      Flags
 1      8389kB  12.6MB  4194kB               uboot
 2      12.6MB  16.8MB  4194kB               trust
 3      16.8MB  21.0MB  4194kB               misc
 4      21.0MB  25.2MB  4194kB               dtbo
 5      25.2MB  41.9MB  16.8MB               resource
 6      41.9MB  83.9MB  41.9MB               kernel
 7      83.9MB  134MB   50.3MB               boot
 8      134MB   747MB   612MB   ext4         rootfs
 9      747MB   63.9GB  63.1GB  ext4

hle5128, 2022-05-12:
Im able to booted into openwrt OS using this build latest version.
https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases

hle5128, 2022-05-12:
no problem with this build:

AO Build@2022.05.01 r16590-13c88950a2

TalalMash, 2022-05-12:
Ok I'll build FriendlyWRT soon.

AO ***@***.*** r16590-13c88950a2

hle5128, 2022-05-12:
I got 10/10 successfully booting from friendlywrt latest version vs. OpenWRT or any other snapshot for rk3399; maybe one success out of 20 successfully tried (reflash, power on/off, SYS on but no IP, etc..)
But recently found this build has a more successful chance of booting than others.
https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds <https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds>

TalalMash, 2022-05-12:
I'll send an installer script soon (install in Terminal/SSH), FriendlyWRT and the AO build are missing many packages required, and mixing OWRT official packages causes conflict. Will release build once the mainline supports all types of SD cards.

hle5128, 2022-05-12:
so the script will just install over on top of any OS (gl inet and friendlywrt) or just friendlywrt?
i have the flint here I could give it a trial as well.
just bought some v30 a2 sd card so i have like 4 of them to test but i dont think its just sdcard, because they are relatively the same, maybe the kernel picky to boot on certain configuration sdcard.

hle5128, 2022-05-12:
What I mean is maybe we can mirror the image of the latest AO build, see where is an original snapshot and use that OpenWrt build? See the kernel version, or has any patch fixed?

[74 more replies]

closed 2022-05-17 by TalalMash