SmoothWAN/SmoothWAN #36: Raspi 4 v0.99.6 disconnect every 30 mins or so

hle5128, 2022-05-15:
Hi,
I left speedify run overnight, and I notice it gets auto disconnected or so on every 30 mins or close to it.
I have two modem 5G hooks up to it, and they are online just fine; the reason I know is I set the speed test on my nas every 30 mins throughout the day, and when speedify auto disconnects, the speed test is still run from one of the two interfaces.

Here are the logs I got in the morning; I turned them on and 30 mins later disconnected again.

Sun May 15 09:50:43 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:50:59 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:51:15 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:51:31 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:51:47 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:52:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:52:19 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:52:35 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:52:51 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:53:07 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:53:23 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:53:39 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:53:55 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:54:11 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:54:27 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:54:43 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:54:59 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:55:15 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:55:31 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:55:47 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:56:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:56:19 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:56:35 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:56:51 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:57:07 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:57:23 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:57:39 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:57:55 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:58:11 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:58:27 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:58:43 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:58:59 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:59:15 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:59:31 2022 daemon.err odhcpd[3114]: 
Sun May 15 09:59:47 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:00:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:00:19 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:00:35 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:00:51 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:01:07 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:01:23 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:01:32 2022 daemon.notice netifd: port4 (3756): udhcpc: sending renew to 192.168.3.1
Sun May 15 10:01:32 2022 daemon.notice netifd: port4 (3756): udhcpc: lease of 192.168.3.83 obtained, lease time 43200
Sun May 15 10:01:39 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:01:55 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:02:11 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:02:27 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:02:43 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:02:47 2022 daemon.notice netifd: por5 (3757): udhcpc: sending renew to 192.168.4.1
Sun May 15 10:02:47 2022 daemon.notice netifd: por5 (3757): udhcpc: lease of 192.168.4.83 obtained, lease time 43200
Sun May 15 10:02:59 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:03:15 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:03:31 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:03:47 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:04:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:04:19 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:04:35 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:04:51 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:05:07 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:05:23 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:05:39 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:05:55 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:06:11 2022 daemon.err odhcpd[3114]: 
Sun May 15 10:06:27 2022 daemon.err odhcpd[3114]: 
[... 445 lines cut ...]
Sun May 15 11:28:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:28:19 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:28:35 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:28:51 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:29:07 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:29:23 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:29:39 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:29:55 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:30:11 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:30:26 2022 daemon.err nginx[4189]: 2022/05/15 11:30:26 [error] 4207#0: *151 open() "/www/spdui/WorkSans-Regular.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Regular.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=829529"
Sun May 15 11:30:26 2022 daemon.err nginx[4189]: 2022/05/15 11:30:26 [error] 4207#0: *152 open() "/www/spdui/WorkSans-SemiBold.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-SemiBold.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=829529"
Sun May 15 11:30:26 2022 daemon.err nginx[4189]: 2022/05/15 11:30:26 [error] 4207#0: *153 open() "/www/spdui/WorkSans-Bold.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Bold.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=829529"
Sun May 15 11:30:26 2022 daemon.err nginx[4189]: 2022/05/15 11:30:26 [error] 4207#0: *154 open() "/www/spdui/WorkSans-Light.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Light.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=829529"
Sun May 15 11:30:27 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:30:43 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:30:59 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:31:15 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:31:31 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:31:47 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:32:03 2022 daemon.err odhcpd[3114]: 
Sun May 15 11:32:08 2022 user.info : luci: accepted login on /admin/status/overview for root from 127.0.0.1
Sun May 15 11:32:09 2022 daemon.err nginx[4189]: 2022/05/15 11:32:09 [error] 4207#0: *167 open() "/www/spdui/WorkSans-Regular.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Regular.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=23936"
Sun May 15 11:32:09 2022 daemon.err nginx[4189]: 2022/05/15 11:32:09 [error] 4207#0: *168 open() "/www/spdui/WorkSans-SemiBold.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-SemiBold.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=23936"
Sun May 15 11:32:09 2022 daemon.err nginx[4189]: 2022/05/15 11:32:09 [error] 4207#0: *170 open() "/www/spdui/WorkSans-Bold.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Bold.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=23936"
Sun May 15 11:32:09 2022 daemon.err nginx[4189]: 2022/05/15 11:32:09 [error] 4211#0: *171 open() "/www/spdui/WorkSans-Light.ttf" failed (2: No such file or directory), client: 127.0.0.1, server: _lan, request: "GET /spdui/WorkSans-Light.ttf HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/spdui/?uid=23936"

TalalMash, 2022-05-15:
Try to connect to "Torrent friendly server" to see if Speedify is disconnecting due to P2P traffic.

TalalMash, 2022-05-15:
To generate Speedify logs, do tar -zcvf /www/spdlog.tar.gz /usr/share/speedify/logs/
Visit http://172.17.17.2/spdlog.tar.gz to download and upload it here.

hle5128, 2022-05-15:
[spdlog.tar.gz](https://github.com/TalalMash/SmoothWAN/files/8695369/spdlog.tar.gz)

here is it

TalalMash, 2022-05-15:
[image]
Torrenting is detected.
You can either use a torrenting server, or another country that has the ability to block torrents as opposed to servers that disconnect, I'm not aware on how to segregate these.
You can also use VPN on top of VPN, it's preferred to set Speedify transport mode to UDP mode and use a UDP VPN.

hle5128, 2022-05-15:
probably I'm going to run the wireguard client on the nas server.
been testing alot lately with smoothwan, and one thing i notice, combining too wans do make speed slower.
For example, each wan individual when i test up to 200 to 300 Mbps but when run through smoothwan, i only get around 150ish max on both, regardless which mode i select or protocol.
I might just be running this through the flint instead just 1 wan and run wireguard client on it. Instead of doing speedify

TalalMash, 2022-05-15:
I included the speeds in the README page for each device, 200-300 each is way too high for a Pi4 to handle, try disabling encryption although the difference will be minimal. The flint will be much slower and the R4S slightly faster. 
x86 hardware may be the only option e.g Intel J3060 Mini PC (~$150+).

TalalMash, 2022-05-15:
You can also discuss the performance for the NanoPi R4S in it's issue page, you can overclock to 2GHz (safe max) to check, there is one testing report left regarding AES hardware acceleration.

hle5128, 2022-05-15:
What about r4s? 
I might het the intel j3060 mini pc to test today.

hle5128, 2022-05-15:
Yes, i coundnt get it online. I posted in it issues, can you take a look. Want to test AES for you since all my 3 wans combined over 1G

TalalMash, 2022-05-15:
> Intel J3060

To be on the safe side, I'd recommend something with more power than this baseline family, try to avoid Braswell series.

TalalMash, 2022-05-15:
> What about r4s? I might het the intel j3060 mini pc to test today.
> […](#)

To be on the safe side, I'd recommend something with more power than this baseline family, try to avoid Braswell series.
An i3 based Qotom should do well, includes multiple NICs as a bonus.

closed 2022-05-17 by TalalMash