SmoothWAN/SmoothWAN #37: SmoothWAN on OWRT-based routers

TalalMash, 2022-05-17:
Installer script included in [pre-release section](https://github.com/TalalMash/SmoothWAN/releases/tag/script).
Issue opened for testing and feedback.
For testing:
- GL.inet Cortex-A53+ routers

TalalMash, 2022-05-17:
@hle5128 You can give this a try on your Flint, check free-space using `df -h` (overlay size)

hle5128, 2022-05-17:
> @hle5128 You can give this a try on your Flint, check free-space using `df -h` (overlay size)

ok let me try, i will try on flint and r4s friendlywrt and will report back shortky

hle5128, 2022-05-18:
> @hle5128 You can give this a try on your Flint, check free-space using `df -h` (overlay size)

this is running on r4s, notice the //usr/lib/opkg/info/speedifyconf.postinst: line 303: /etc/init.d/speedifyconf: Permission denied at the end ?

root@FriendlyWrt:~# sh swarmsetup.sh

╔═╗┌┬┐┌─┐┌─┐┌┬┐┬ ┬╦ ╦╔═╗╔╗╔
╚═╗││││ ││ │ │ ├─┤║║║╠═╣║║║
╚═╝┴ ┴└─┘└─┘ ┴ ┴ ┴╚╩╝╩ ╩╝╚╝
[Minimal Install]
 * Starting install in 5 seconds, press CTRL-C to cancel
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Package unzip (6.0-8) installed in root is up to date.
Installing nginx-mod-luci (1.21.3-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/nginx-mod-luci_1.21.3-1_aarch64_generic.ipk
Installing uwsgi (2.0.20-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/uwsgi_2.0.20-1_aarch64_generic.ipk
Installing uwsgi-syslog-plugin (2.0.20-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/uwsgi-syslog-plugin_2.0.20-1_aarch64_generic.ipk
Installing uwsgi-cgi-plugin (2.0.20-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/uwsgi-cgi-plugin_2.0.20-1_aarch64_generic.ipk
Installing uwsgi-luci-support (2.0.20-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/uwsgi-luci-support_2.0.20-1_aarch64_generic.ipk
Installing nginx-util (1.6-14) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/nginx-util_1.6-14_all.ipk
Installing nginx-ssl-util (1.6-14) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/nginx-ssl-util_1.6-14_aarch64_generic.ipk
Installing nginx-ssl (1.21.3-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/nginx-ssl_1.21.3-1_aarch64_generic.ipk
Installing nginx (1.21.3-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/nginx_1.21.3-1_all.ipk
Package luci-compat (git-22.069.45071-03bb0e2) installed in root is up to date.
Package curl (7.82.0-2) installed in root is up to date.
Installing wget-ssl (1.21.2-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/wget-ssl_1.21.2-1_aarch64_generic.ipk
Installing ar (2.37-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/base/ar_2.37-2_aarch64_generic.ipk
Installing libbfd (2.37-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/base/libbfd_2.37-2_aarch64_generic.ipk
Installing luci-app-vpnbypass (1.3.1-9) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/luci/luci-app-vpnbypass_1.3.1-9_all.ipk
Installing vpnbypass (1.3.2-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/packages/vpnbypass_1.3.2-1_all.ipk
Package ar (2.37-2) installed in root is up to date.
Installing binutils (2.37-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/aarch64_generic/base/binutils_2.37-2_aarch64_generic.ipk
Installing libopcodes (2.37-2) to root...
[... 737 lines cut ...]
  inflating: packages/luci-app-tailscaleconf_1.0_all.ipk  
Installing luci-app-speedifyconf (1.2) to root...
Installing speedifyconf (4.4) to root...
Upgrading luci-app-vpnbypass on root from 1.3.1-9 to 1.3.4-1...
[... cut ...]
Error: any valid address is expected rather than "".
vpnbypass 1.3.5-1 started with TID: 200; FW_MARK: 0x010000
vpnbypass 1.3.5-1 monitoring interfaces: 
Configuring luci-app-vpnbypass.
Configuring speedifyconf.
//usr/lib/opkg/info/speedifyconf.postinst: line 301: /etc/init.d/speedifyconf: Permission denied
//usr/lib/opkg/info/speedifyconf.postinst: line 303: /etc/init.d/speedifyconf: Permission denied
Configuring luci-app-speedifyconf.
Rebooting in 10 seconds, press CTRL-C to cancel
root@FriendlyWrt:~# 
Session was closed

hle5128, 2022-05-18:
successful install speedify but app wont recognize that is online.

root@FriendlyWrt:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd00:ab:cd::/48'

config device
        option name 'eth0'
        option macaddr '80:34:28:38:77:7e'

config interface 'wan'
        option device 'eth0'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config device
        option name 'eth1'
        option macaddr '82:34:28:38:77:7e'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.2.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'docker0'

root@FriendlyWrt:~# 

root@FriendlyWrt:~# ip route
default via 192.168.3.1 dev eth0 proto static src 192.168.3.2 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
192.168.2.0/24 dev br-lan proto kernel scope link src 192.168.2.1 
192.168.3.0/24 dev eth0 proto kernel scope link src 192.168.3.2 

[image]
[image]

hle5128, 2022-05-18:
here is the log
[spdlog.tar.gz](https://github.com/TalalMash/SmoothWAN/files/8712414/spdlog.tar.gz)

hle5128, 2022-05-18:
> @hle5128 You can give this a try on your Flint, check free-space using `df -h` (overlay size)

the size on flint, is only 60.7M, does it enough?

ilesystem                Size      Used Available Use% Mounted on
mtd:ubi_rootfs           30.8M     30.8M         0 100% /rom
tmpfs                   210.3M    328.0K    210.0M   0% /tmp
/dev/ubi0_3              60.7M      4.0M     53.5M   7% /overlay
overlayfs:/overlay       60.7M      4.0M     53.5M   7% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mtdblock12           2.0M      2.0M         0 100% /lib/firmware/IPQ6018/WIFI_FW
root@GL-AX1800:~#

hle5128, 2022-05-18:
> successful install speedify but app wont recognize that is online.
> 
> root@FriendlyWrt:~# cat /etc/config/network
> 
> config interface 'loopback' option device 'lo' option proto 'static' option ipaddr '127.0.0.1' option netmask '255.0.0.0'
> 
> config globals 'globals' option ula_prefix 'fd00🆎cd::/48'
> 
> config device option name 'eth0' option macaddr '80:34:28:38:77:7e'
> 
> config interface 'wan' option device 'eth0' option proto 'dhcp'
> 
> config interface 'wan6' option device 'eth0' option proto 'dhcpv6'
> 
> config device option name 'br-lan' option type 'bridge' list ports 'eth1'
> 
> config device option name 'eth1' option macaddr '82:34:28:38:77:7e'
> 
> config interface 'lan' option device 'br-lan' option proto 'static' option ipaddr '192.168.2.1' option netmask '255.255.255.0' option ip6assign '60'
> 
> config interface 'docker' option device 'docker0' option proto 'none' option auto '0'
> 
> config device option type 'bridge' option name 'docker0'
> 
> root@FriendlyWrt:~#
> 
> root@FriendlyWrt:~# ip route default via 192.168.3.1 dev eth0 proto static src 192.168.3.2 172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 192.168.2.0/24 dev br-lan proto kernel scope link src 192.168.2.1 192.168.3.0/24 dev eth0 proto kernel scope link src 192.168.3.2
> 
> [image]
> 
> [image]

root@FriendlyWrt:/usr/share/speedify# ifconfig
br-lan    Link encap:Ethernet  HWaddr 82:34:28:38:77:7E  
          inet addr:192.168.2.1  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3617 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3763 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:756271 (738.5 KiB)  TX bytes:3117436 (2.9 MiB)

connectify0 Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.202.0.2  P-t-P:10.202.0.2  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:14800  Metric:1
          RX packets:17 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000 
          RX bytes:7227 (7.0 KiB)  TX bytes:8355 (8.1 KiB)

docker0   Link encap:Ethernet  HWaddr 02:42:F6:D0:77:27  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 80:34:28:38:77:7E  
          inet addr:192.168.3.2  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: fe80::8234:28ff:fe38:777e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1766 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1178542 (1.1 MiB)  TX bytes:260466 (254.3 KiB)
          Interrupt:31 

eth1      Link encap:Ethernet  HWaddr 82:34:28:38:77:7E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3660 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3772 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:809807 (790.8 KiB)  TX bytes:3117702 (2.9 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:15920 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15920 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1787360 (1.7 MiB)  TX bytes:1787360 (1.7 MiB)

i can see interface connectify0

hle5128, 2022-05-18:
> here is the log [spdlog.tar.gz](https://github.com/TalalMash/SmoothWAN/files/8712414/spdlog.tar.gz)
> 
> also, [image] i notice there is no speedify interfaces?

i created manually interface attach connectify0 then start speedify_cli manaually, everything work under command, so far no problem or crash for over hours. But Speedify GUI is not working.

TalalMash, 2022-05-18:
> Yeah, networking is done manually such as adding 'connectify0' since configuration is unpredictable between devices. I'll add instructions in the downloads per device if it works.
> I'll add extroot step in another script for the Flint, you will need any USB flash drive plugged in to expand storage, it also seems to run older OWRT.
> For the Nano, do `cat /etc/firewall.custom`
>
> The page is getting cluttered, you can save each log result in a text and upload here instead of pasting.

To clear cache of the current page in Chrome on Mac: Command+Shift+R
Use it on the Nano Speedify UI Overview page to double check.

also, [Screen Shot 2022-05-17 at 8 54 15 PM] <https://user-images.githubusercontent.com/8794968/168935404-778bdda5-f2a0-473d-a017-7c9e16bde366.png>  i notice there is no speedify interfaces?

TalalMash, 2022-05-18:
Updated the pre-release and added Wiki info.
Regarding the Nano post:
- `cat /etc/firewall.custom`
- `netstat -tulpn | grep LISTEN`

hle5128, 2022-05-18:
Flint can not install, have you checked the logs?

TalalMash, 2022-05-18:
Have you tried the new script? Yes.. I did check the logs for the older script.
Have you read the Wiki? There is a step to expand storage.

TalalMash, 2022-05-18:
In case you don't know where the Wiki section is:
[image]

hle5128, 2022-05-18:
Hi,
Will give it a try, and report back on brume and flint

TalalMash, 2022-05-18:
You may need to install LuCi from the GUI first, added in Wiki.

hle5128, 2022-05-19:
> You may need to install LuCi from the GUI first, added in Wiki.

got some errors:
extroot.sh: line 18: block: not found
NTFS signature is missing.
Failed to mount '/dev/sda1': Invalid argument
The device '/dev/sda1' doesn't seem to have a valid NTFS.
Maybe the wrong device is used? Or the whole disk instead of a
partition (e.g. /dev/sda, not /dev/sda1)? Or the other way around?

https://share.getcloudapp.com/jkuXd58X

also, after reboot, system is dead, would not given any ip out, manually set ip wont connect to the system.

hle5128, 2022-05-19:
> You may need to install LuCi from the GUI first, added in Wiki.

i can unplug the usb and mount on my system and see the files there but the flint wont boot after reboot of extroot.sh

hle5128, 2022-05-19:
i skipped the step usb since it doesnt boot, run swarm-chaos.sh and still got this error at the end:
i dont think it found lib "libuci20130104" to install,

Configuring luci-app-vpnbypass.
Configuring speedifyconf.
//usr/lib/opkg/info/speedifyconf.postinst: line 5: /etc/init.d/speedifyconf: Permission denied
//usr/lib/opkg/info/speedifyconf.postinst: line 5: /etc/init.d/speedifyconf: Permission denied
Configuring luci-app-speedifyconf.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-speedifyconf:
 *      libuci20130104
Note: Set Speedify firewall to RED zone
Rebooting in 10 seconds, press CTRL-C to cancel

hle5128, 2022-05-19:
root@GL-AX1800:/usr/share/speedify# ls -asl
     0 drwxrwxr-x    3 root     root           976 May 19 23:35 .
     0 drwxrwxr-x    1 root     root           768 May 19 23:35 ..
     4 -rwxrwxr-x    1 root     root           303 May 12 15:43 DisableRpFilter.sh
     4 -rwxrwxr-x    1 root     root          1928 May 12 15:43 GenerateLogs.sh
    12 -rwxrwxr-x    1 root     root         11647 May 12 15:43 ShareSpeedify.sh
   100 -rw-rw-r--    1 root     root         98646 May 12 15:43 SpeedifyCLI.pdf
     4 -rwxrwxr-x    1 root     root           528 May 12 15:43 SpeedifyShutdown.sh
     4 -rwxrwxr-x    1 root     root          1227 May 12 15:43 SpeedifyStartup.sh
   204 -rw-rw-r--    1 root     root        208065 May 12 15:43 cacert.pem
     4 -rw-rw-r--    1 root     root          1270 May 12 15:43 default.conf
     0 drwxr-xr-x    2 root     root           160 May 19 23:35 logs
  7952 -rwxrwxr-x    1 root     root       8142768 May 12 15:43 speedify
  1284 -rwxrwxr-x    1 root     root       1314360 May 12 15:43 speedify_cli
root@GL-AX1800:/usr/share/speedify# ./speedify_cli 
./speedify_cli: line 2: syntax error: unterminated quoted string
[image]

TalalMash, 2022-05-19:
Getting closer, its quite tedious to predict things remotely, I need to get my hands on one or remote desktop to make it trivial.

[Screen Shot 2022-05-19 at 7 48 01 PM]<https://user-images.githubusercontent.com/8794968/169421990-e37f3b7c-e041-4cc2-8c66-69d0a72bb288.png>

hle5128, 2022-05-20:
we can setup a time to do teamviewer
let me know your time zone

hle5128, 2022-05-20:
lets setup a time, i can hope online and let you do remotely, we can debug on both flint and r4s

TalalMash, 2022-05-20:
I'm available on Speedify's Discord SmoothWAN channel or DM.
Will post progress here for reference.

hle5128, 2022-05-20:
what is your link channel?

TalalMash, 2022-05-20:
The invite link is in the README top

hle5128, 2022-05-20:
not in here 
https://github.com/TalalMash/SmoothWAN/blob/main/README.md <https://github.com/TalalMash/SmoothWAN/blob/main/README.md>

TalalMash, 2022-05-20:
That's odd:
[image]
Here's the direct link: https://discord.gg/AxSSjpgwjx

TalalMash, 2022-05-22:
Armv7l-e library incompatible, porting next version.

closed 2022-05-22 by TalalMash

hle5128, 2022-07-02:
Just want to give you an updated.
I just got the new version slate arrived today. Do you have any built i can test?

Thanks

TalalMash, 2022-07-02:
> Just want to give you an updated. I just got the new version slate arrived today. Do you have any built i can test? Thanks
> […](#)

Mine didn't arrive yet, "late July".

hle5128, 2022-07-04:
Should we try the script?

TalalMash, 2022-07-08:
@hle5128 Good news! Flint and Brume/W beta firmware is now using OpenWRT 21.02. Slate AX / Flint / Brume will use the same package. 
Added in the new issue page.

hle5128, 2022-07-19:
any update on your device?

hle5128, 2022-07-26:
anyupdate on the customs :)