SmoothWAN/SmoothWAN #41: Support for GL.iNet Slate AX - No dongles or access point needed

TalalMash, 2022-06-03:
**Status: Pre-order**
_WIP_

hle5128, 2022-07-05:
Here is for more information.
[image]
[image]

TalalMash, 2022-07-05:
Looks promising! Mine will arrive in a week or so, there will be no script,
only a simple file to upload and it will automate the installation and
configuration.

psylligent, 2022-07-24:
@TalalMash , amazing work. Will you make smoothwan available for GL-AX1800, I have speedify paid for a year, and 3x different connections, VDSL, 5g via a tether  and 4g usb modem. I already bought the router so I'm happy to beta test for you. If smoothwan is not supported on this model,I'll return and buy the one you recommend. Really love your work , so looking forward to seeing the end result. Thanks

TalalMash, 2022-07-24:
@psylligent Thank you, I'm still waiting for the delivery arrival. Fortunately the GL-AX1800 beta (4.0+) is the same firmware and SoC as the Slate AX, it should work with little tweaks.
Will comment here on release to test it

psylligent, 2022-07-24:
@TalalMash amazing, can't wait to try. Hats of for coming up with this solution.
Onc
Will send you Chai late mate bro!

psylligent, 2022-07-31:
> @TalalMash , amazing work. Will you make smoothwan available for GL-AX1800, I have speedify paid for a year, and 3x different connections, VDSL, 5g via a tether  and 4g usb modem. I already bought [...]

@TalalMash can't wait to give it a try,  I've actually returned the GL-AX1800 Flint and ordered GL-AXT1800 (Slate AX), the Flint was very unstable with the v4 beta. Can't wait bro 😂

TalalMash, 2022-08-01:
Status update: device delivered, currently in development.

TalalMash, 2022-08-06:
First image released.

hle5128, 2022-08-06:
Will test tonight and report.
Speed performance vs ras pi 4?

TalalMash, 2022-08-07:
> Will test tonight and report. Speed performance vs ras pi 4?
> […](#)

It measured roughly around 210Mbit/s with/without encryption enabled, compared to Pi 160Mbit/s with encryption, 350Mbit/s without encryption.
Note for readers, use a mini-pc/laptop for 200Mbit/s+ as the latency is bad with the Pi above this point.

hle5128, 2022-08-10:
got it up and running, but which port is for the LAN? 
LAN2 + WAN is for WAN1/WAN2 right? so is LAN1 is for LAN?

TalalMash, 2022-08-10:
> got it up and running, but which port is for the LAN? LAN2 + WAN is for WAN1/WAN2 right? so is LAN1 is for LAN?

The diagram is in the README page in Quick Setup.

psylligent, 2022-08-10:
> Will test tonight and report.
> Speed performance vs ras pi 4?
> 
> 
> > 
> > 
> > First image released.

Sorry if I'm thick, how did you flash on Slate ?

psylligent, 2022-08-10:
My setup:

Internet connections available 

ISP router, connect to slate via eth (73Mb)
Slate will connect to 5G router via WiFi (5ghz) (approx 500 Mb)
4g usb dongle plugged in slate usb port (approximately 150Mb)

I'm soo curious to see what the combined total speed will become available via speedify, and if it will truly combine the above to a total of 723 Mb 🤣

TalalMash, 2022-08-10:
> My setup:
>
> Internet connections available
>
> ISP router, connect to slate via eth (73Mb) Slate will connect to 5G router via WiFi (5ghz) (approx 500 Mb) 4g usb dongle plugged in slate usb port [...]

I'd suggest a Tp-link ER605 load balancer considering that you have stable connections and very high single conn. throughput from one of the connections.
AFAIK Speedify public servers hover around 400-700 from my tests and there is FUP, it doesn't target setups with really good/fast connections from my experience, unless of coarse you need VPN for privacy or combining a really asymmetric connection to boost upload, e.g Comcast offering 1.2Gbit down 40Mbit up. Then a PC and a private server plan would work for gigabit+ according to a Speedify live stream. 

The Slate AX will definitely not do more than 250 Mbit.

But if you have another much slower/unstable internet setup, checkout the expandable section in Releases download page, it's a file that you upload in GLiNet's Upgrade page, the device will reboot and broadcast an access point, then follow the quick setup section screenshots in the README page.

psylligent, 2022-08-11:
Super and thanks, I’ll look into it 😊

TalalMash, 2022-08-14:
@nickpoorman 
Nice setup! Good to hear its working, I haven't tested the max current limit on the USB port so this is good news.
The power supply is rated at 20W, the max I could measure on the Slate AX is 8W so there is a plenty of headroom.
I contacted GL.iNet regarding few details and there is an MediaTek based equivalent to the Slate AX with 5G modem upcoming, hopefully with one/two more ports. 

A small tip for readers you can name your VLANs to appear on Speedify from the "Devices" tap:
[image]
Will add a VLAN guide in the Wiki.

psylligent, 2022-08-15:
> Loaded the firmware up on my Slate AX. Everything seems to be working great. I attached a powered USB hub to the Slate USB port (it's really a PiHat--Waveshare USB 3.2 Gen1 HUB/Gigabit ETH HAT) and plugged in a Atheros AR9271 along with a 5G T-Mobile USB MiFi device into the hat USB3 ports.
> 
> I have everything running through eth0 with a single managed TP Link switch with VLANS. Upstream I have Starlink, a Pepwave router, and an Alfa CampPro 3. Everything is bonded and running well so far.
> 
> [image]
> 
> [image]
> 

Looks good, what usb hub did u use ?
Also, what's the max speed your getting? 
I get max 120Mbps in UK via speedify, but 250Mbps when connected to 5g only

psylligent, 2022-08-15:
> @nickpoorman
> Nice setup! Good to hear its working, I haven't tested the max current limit on the USB port so this is good news.
> The power supply is rated at 20W, the max I could measure on the [...]

Thanks @TalalMash 
Any chance you can make it work with nextdns ? There's a way to call the next dns API and update the IP every minute.

Next dns is so much better than adguardhome, they work really well together .

Thanks

TalalMash, 2022-08-16:
I'll look into it, for now it will be a package you upload in the Web UI.

psylligent, 2022-08-16:
> I'll look into it, for now it will be a package you upload in the Web UI.
> 
> 
> > @nickpoorman <https://github.com/nickpoorman>
> > Nice setup! Good to hear its working, I haven't tested the max current
> > limit on the USB port so this is good news.
> > The power supply is rated at 20W, the max I could measure on the Slate AX
> > is 8W so there is a plenty of headroom.
> > I contacted GL.iNet regarding few details and there is an MediaTek based
> > equivalent to the Slate AX with 5G modem upcoming, hopefully with one/two
> > more ports.
> >
> > A small tip for readers you can name your VLANs to appear on Speedify from
> > the "Devices" tap:
> > [image: image]
> > <https://user-images.githubusercontent.com/96490382/184552507-08cb4661-0f9c-4a6c-bb73-ed1d31de749c.png>
> > Will add a VLAN guide in the Wiki.
> >
> > Thanks @TalalMash <https://github.com/TalalMash>
> > Any chance you can make it work with nextdns ? There's a way to call the
> > next dns API and update the IP every minute.
> >
> > Next dns is so much better than adguardhome, they work really well
> > together .
> >
> > Thanks

Thanks, that should be perfect and good enough.

Like speedify, next dns can be loaded via wrap , see https://play.google.com/store/apps/details?id=com.doubleangels.nextdnsmanagement

Thanks for considering

psylligent, 2022-08-16:
> Status update: device delivered, currently in development.

Wanted to feedback, installed, followed all the steps, one thing is not working (it's possible I'm missing something out here) the 5ghz radio, when clicking scan , there's no WiFi network, the scanning stops after 3 seconds. Was only able to scan and connect as client on 2.4ghz, obviously link speed is not as good. I have factory rebooted twice and same result, please,if I'm doing anything wrong let me know 😉

TalalMash, 2022-08-16:
> Wanted to feedback, installed, followed all the steps, one thing is not working (it's possible I'm missing something out here) the 5ghz radio, when clicking scan , there's no WiFi network, the [...]

I've noticed the same, simply disable the AP before scanning, added to Wiki troubleshooting.

TalalMash, 2022-09-10:
Flint build released for testing.

faroit, 2022-09-18:
Nice project. I have a slate ax and wonder if this also works with a tethered android usb wan?

TalalMash, 2022-09-18:
> Nice project. I have a slate ax and wonder if this also works with a tethered android usb wan?

Thanks, yes it does, note that iOS tethering is not reliable.

nickpoorman, 2022-09-20:
Has anyone been able to use radio0 or radio1 as an AP? What settings are you using? I've tried a ton of different combinations but when I try to connect with my iPhone it says "Unable to join network". This is what I'm seeing in the logs:

```
Tue Sep 20 00:16:25 2022 daemon.notice netifd: Wireless device 'radio1' is now up
Tue Sep 20 00:16:26 2022 daemon.info [2628]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/soc/c000000.wifi': not supported by any plugin
Tue Sep 20 00:16:29 2022 kern.warn kernel: [ 1379.828831] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
Tue Sep 20 00:16:29 2022 kern.err kernel: [ 1380.048796] usb 1-1.2: device not accepting address 86, error -62
Tue Sep 20 00:16:30 2022 kern.info kernel: [ 1380.198795] usb 1-1.2: new high-speed USB device number 87 using xhci-hcd
Tue Sep 20 00:16:35 2022 kern.warn kernel: [ 1385.268598] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
Tue Sep 20 00:16:35 2022 daemon.notice hostapd: wlan1: ACS-COMPLETED freq=2437 channel=6
Tue Sep 20 00:16:35 2022 daemon.notice hostapd: wlan1: interface state ACS->HT_SCAN
Tue Sep 20 00:16:37 2022 daemon.err collectd[3292]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/SmoothWAN/sensors-ath11k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1663632998 when last update time is 1663632998 (minimum one second step)
Tue Sep 20 00:16:37 2022 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=6 sec=10 based on overlapping BSSes
Tue Sep 20 00:16:37 2022 daemon.notice hostapd: Fallback to 20 MHz
Tue Sep 20 00:16:38 2022 kern.info kernel: [ 1388.136563] br-lan: port 3(wlan1) entered blocking state
Tue Sep 20 00:16:38 2022 kern.info kernel: [ 1388.136596] br-lan: port 3(wlan1) entered forwarding state
Tue Sep 20 00:16:38 2022 daemon.notice netifd: Network device 'wlan1' link is up
Tue Sep 20 00:16:38 2022 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Tue Sep 20 00:16:38 2022 daemon.notice hostapd: wlan1: AP-ENABLED
Tue Sep 20 00:16:40 2022 kern.warn kernel: [ 1390.548377] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
Tue Sep 20 00:16:40 2022 kern.err kernel: [ 1390.768326] usb 1-1.2: device not accepting address 87, error -62
Tue Sep 20 00:16:40 2022 kern.err kernel: [ 1390.768555] usb 1-1-port2: unable to enumerate USB device
Tue Sep 20 00:16:40 2022 kern.info kernel: [ 1390.918322] usb 1-1.4: new high-speed USB device number 88 using xhci-hcd
Tue Sep 20 00:16:46 2022 kern.err kernel: [ 1396.138112] usb 1-1.4: device descriptor read/64, error -110
Tue Sep 20 00:16:58 2022 daemon.notice hostapd: wlan1: STA fe:69:f2:f5:05:ec IEEE 802.11: did not acknowledge authentication response
Tue Sep 20 00:16:58 2022 daemon.notice hostapd: wlan1: STA fe:69:f2:f5:05:ec IEEE 802.11: did not acknowledge authentication response
Tue Sep 20 00:16:59 2022 daemon.notice hostapd: wlan1: STA fe:69:f2:f5:05:ec IEEE 802.11: did not acknowledge authentication response
Tue Sep 20 00:16:59 2022 daemon.notice hostapd: wlan1: STA fe:69:f2:f5:05:ec IEEE 802.11: did not acknowledge authentication response
Tue Sep 20 00:16:59 2022 daemon.notice hostapd: wlan1: STA fe:69:f2:f5:05:ec IEEE 802.11: did not acknowledge authentication response
Tue Sep 20 00:17:01 2022 kern.err kernel: [ 1411.497472] usb 1-1.4: device descriptor read/64, error -110
Tue Sep 20 00:17:01 2022 kern.info kernel: [ 1411.777441] usb 1-1.4: new high-speed USB device number 89 using xhci-hcd
Tue Sep 20 00:17:06 2022 kern.err kernel: [ 1416.947248] usb 1-1.4: device descriptor read/64, error -110
Tue Sep 20 00:17:07 2022 daemon.err collectd[3292]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/SmoothWAN/sensors-ath11k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1663633028 when last update time is 1663633028 (minimum one second step)
Tue Sep 20 00:17:22 2022 kern.err kernel: [ 1432.376621] usb 1-1.4: device descriptor read/64, error -110
Tue Sep 20 00:17:22 2022 kern.info kernel: [ 1432.496803] usb 1-1-port4: attempt power cycle
Tue Sep 20 00:17:24 2022 kern.info kernel: [ 1434.086531] usb 1-1.4: new high-speed USB device number 90 using xhci-hcd
Tue Sep 20 00:17:29 2022 kern.warn kernel: [ 1439.106350] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
```

TalalMash, 2022-09-20:
@nickpoorman , does it happen with non-iOS clients? Here are the optimal parameters:

- 2.4GHz
[image]
[image]
[image]
[image]
- 5Ghz
[image]
[image]
[image]
[image]

hle5128, 2022-10-20:
Hi,

Do you have support for Slate AX on openmptcp? 
want to test them out but couldnt find the support
i can only find the tree here:
https://github.com/suyuan168/3grouter/tree/ipq60xx <https://github.com/suyuan168/3grouter/tree/ipq60xx>

TalalMash, 2022-10-29:
> Hi, Do you have support for Slate AX on openmptcp? want to test them out but couldnt find the support i can only find the tree here: https://github.com/suyuan168/3grouter/tree/ipq60xx <https://github.com/suyuan168/3grouter/tree/ipq60xx>
> […](#)

This is irrelevant but to answer your question that tree is not APNoS from Qualcomm (+kernel) as far as I know, Wi-Fi and NSS among other things will not work.

TalalMash, 2022-10-31:
No, closing issue, major functions resolved.

closed 2022-10-31 by TalalMash