SmoothWAN/SmoothWAN #54: Slate AX Custom Bypasses Failed

hle5128, 2022-08-10:
I recorded a video here:

https://share.getcloudapp.com/DOudpyr8

Also, click download log, and config fails.
[image]

TalalMash, 2022-08-10:
I didn't update the app with the current nginx root level, it's different from the Pi, after clicking the button, visit 172.17.17.2/spdlogs.tar.gz
Ill release a hot fix soon.

TalalMash, 2022-08-10:
Regarding the bypass app, I don't have access to my Mac, does it work with Chrome/FF?

TalalMash, 2022-08-10:
Fixed with HR1 release, reopen if the issue persists.
Thanks for the report.

closed 2022-08-10 by TalalMash