SmoothWAN/SmoothWAN #55: Slate AX | error Code 3841

hle5128, 2022-08-10:
Keep getting these logs every second.

```
Wed Aug 10 01:28:21 2022 daemon.info uwsgi-luci: nice: can't execute 'ssd': No such file or directory
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: ["error",
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: {      "errorCode" : 3841,     "errorMessage" : "Timeout",     "errorType" : "Timeout waiting for result" }
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: ]
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is enabled
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Network device 'connectify0' link is up
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' has link connectivity
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is setting up now
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is now up
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain test
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain onion
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain bind
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain lan
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 10.202.0.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.5.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.1.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain test
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain onion
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain bind
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain lan
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 10.202.0.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.5.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.1.1#53
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: add network interface connectify0: event processed
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: event reported: action=add, name=connectify0, subsystem=net
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Aug 10 01:29:16 2022 user.notice firewall: Reloading firewall due to ifup of Speedify (connectify0)
Wed Aug 10 01:26:49 2022 daemon.info uwsgi-luci:
Wed Aug 10 01:28:18 2022 daemon.notice netifd: Network device 'connectify0' link is down
Wed Aug 10 01:28:18 2022 daemon.notice netifd: Interface 'Speedify' has link connectivity loss
Wed Aug 10 01:28:18 2022 daemon.notice netifd: Interface 'Speedify' is now down
Wed Aug 10 01:28:18 2022 daemon.notice netifd: Interface 'Speedify' is disabled
Wed Aug 10 01:28:18 2022 user.notice ModemManager: hotplug: remove network interface connectify0: event processed
Wed Aug 10 01:28:18 2022 user.notice ModemManager: hotplug: event reported: action=remove, name=connectify0, subsystem=net
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain test
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain onion
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain bind
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain lan
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.5.1#53
Wed Aug 10 01:28:18 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.1.1#53
Wed Aug 10 01:28:21 2022 daemon.info uwsgi-luci: nice: can't execute 'ssd': No such file or directory
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: ["error",
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: {      "errorCode" : 3841,     "errorMessage" : "Timeout",     "errorType" : "Timeout waiting for result" }
Wed Aug 10 01:28:25 2022 daemon.info uwsgi-luci: ]
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is enabled
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Network device 'connectify0' link is up
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' has link connectivity
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is setting up now
Wed Aug 10 01:29:15 2022 daemon.notice netifd: Interface 'Speedify' is now up
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain test
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain onion
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
[... 153 lines cut ...]
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain bind
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain lan
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 10.202.0.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.5.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.1.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain test
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain onion
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain localhost
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain local
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain invalid
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain bind
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using only locally-known addresses for domain lan
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 10.202.0.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.5.1#53
Wed Aug 10 01:29:15 2022 daemon.info dnsmasq[3506]: using nameserver 192.168.1.1#53
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: add network interface connectify0: event processed
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: event reported: action=add, name=connectify0, subsystem=net
Wed Aug 10 01:29:16 2022 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Aug 10 01:29:16 2022 user.notice firewall: Reloading firewall due to ifup of Speedify (connectify0)

```

TalalMash, 2022-08-10:
Your Speedify app keeps restarting, tap the generate log button and upload here. Review the log content if /etc/network contains passwords before uploading.

TalalMash, 2022-08-10:
Looking at the log again I think you accidentally copy and pasted the log 3-4 times. That error is normal, it's related to the OLED app on Speedify start up which doesn't exist on the Slate AX. I don't see the bug or issue you're having.

closed 2022-08-10 by TalalMash

TalalMash, 2022-08-10:
Removed the error from the log by few checks in 0.99.8HF1
Thanks for the report.