SmoothWAN/SmoothWAN #58: Per Upload / download bypass options

psylligent, 2022-08-16:
Just had a thought, would it be possible to only route Upload traffic through Speedify, but have the option to mark devices / Mac address, to bypass download ?

So where one ISP gives me 500 Mbs download speed, if I route through Speedify, speed will be reduced heavily based on speedify available top speed (think the maximum I ever got was 200Mbs via an UK server). However, same ISP upload is low on itself (max 20 Mbs). Where, via speedify VPN, 3x combined ISP gives me 40-60Mbps 

It would make sense to be able to specify  which traffic to be routed ?
Thanks

TalalMash, 2022-08-16:
It's possible using a tunnel or Wireguard, firewall configuration and asymmetric routing on both ends (with a VPS) . Your download traffic will still have to go through and exit via the same IP:
- Download: Wireguard over fastest connection 
- Upload: Wireguard over speedify
I haven't tried this with Wireguard yet, will let you know and add it to discussions post. 

The overhead is large here so it would make sense to use a PC. 
Have you tried having only one primary with the fastest connection? It won't try to use the other connections when the speed difference is large which can slow things down, but it your case it will on upload. (Secondary is boost mode)

psylligent, 2022-08-16:
> It's possible using a tunnel or Wireguard, firewall configuration and asymmetric routing on both ends (with a VPS) . Your download traffic will still have to go through and exit via the same IP:
> - [...]

Sounds like a good starting point 
Wireguard on a local pc can make sense, running on same router would be best 
I did try the option with primary only,but speedify is reducing download speed to what their servers can provide. Speedify sells itself as good for streaming, they should give the option to only aggregate uploaded, or both... It would make sense 😉