SmoothWAN/SmoothWAN #69: Support for GL.iNet Flint

TalalMash, 2022-09-19:
Status: Release added - Test version

**TODO:**
- Basic Wi-Fi setup page (5GHz scan workaround) **
- LAN4/WAN4 port switch
- WAN ports order numbers and testing

**Vendor quirks:**
- ** Disable Wi-Fi before clicking the Scan button to scan as a client

DalSoft, 2022-11-03:
I have SmoothWAN working on my GL.iNet Flint. It's been running perfectly using the WAN port (eth0) and port 1 (eth1) as another WAN.

TalalMash, 2022-11-04:
@DalSoft great to hear, I'm guessing the LAN port is eth4?

meavydev, 2022-12-08:
I'm guessing that the GLi.Net Flint image is only compatible with the AXT1800 not the AX1800?
Any suggestions on getting it working ?
[image]

TalalMash, 2022-12-08:
> I'm guessing that the GLi.Net Flint image is only compatible with the AXT1800 not the AX1800? Any suggestions on getting it working ? [image]

It shows Slate in your screenshot, the Flint is in the download's expand section, Github UI may be confusing at times.

meavydev, 2022-12-08:
You're right that I had the wrong file.
Unfortunately it is showing the same error...
[image]
Could it be something has changed in 4.1 in the Flint?

meavydev, 2022-12-08:
This is what shows in System Info
[image]

TalalMash, 2022-12-08:
Did they remove the force upgrade option?

meavydev, 2022-12-08:
> Did they remove the force upgrade option?

I don't see any mention of a force upgrade on their web page: [https://blog.gl-inet.com/step-by-step-guideline-how-to-upgrade-to-the-latest-firmware/](url)

I think they only show the Install button after it has passed verification and you can choose if you want to keep settings / packages before clicking it.
I only got the Flint today, so I am far from expert in its UI...

meavydev, 2022-12-08:
Is it worth trying via the UBoot option?
[https://docs.gl-inet.com/en/3/tutorials/debrick/](url)

TalalMash, 2022-12-08:
> Is it worth trying via the UBoot option? https://docs.gl-inet.com/en/3/tutorials/debrick/

The safer alternative option is to simply upgrade via LuCI!
Head to Advanced settings and click on the link, login and then go to System -> Flash firmware and click Flash image and follow through.

It seems like you are on an old version or so, I'm not sure, but the LuCI option is consistent.

meavydev, 2022-12-09:
I tried Luci, but it seems to fail
[image]
I unchecked the keep and checked force, but it never booted.
However using Uboot has worked and it has just shown SmoothWAN Setup. So I can get to try setting up now.

TalalMash, 2022-12-09:
> I tried Luci, but it seems to fail [image] I unchecked the keep and checked force, but it never booted. However using Uboot has worked and it has just shown SmoothWAN Setup. So I can get to try setting up now.

That's interesting to know thanks for the follow up, will check the image in the next release.

meavydev, 2022-12-09:
I've got Speedify running OK now, but I've got a question about port configuration.
I only want to have two active WAN ports, so WAN for LTE and LAN1 for Starlink.
How do I set the rest of the ports to be LAN rather than used by Speedify?

TalalMash, 2022-12-09:
> I've got Speedify running OK now, but I've got a question about port configuration. I only want to have two active WAN ports, so WAN for LTE and LAN1 for Starlink. How do I set the rest of the ports to be LAN rather than used by Speedify?

I'm adding a quick switch UI in the next release, in the meantime you'd have to go to _Network -> Interface -> Devices (tab)_ and click on _Configure_ next to `br-lan` , you can then add or remove bridge ports for LAN access. 
You will also have to remove the extra WANs in the Interfaces tab by clicking _Delete_ next to the network.

meavydev, 2022-12-09:
Thanks.
Seems to be working now.
Of course various streaming services don't like the Speedify VPN (iPlayer / Apple TV / Amazon), so I have had to setup VPN policies to route the smart TV directly to one of the WANs. 
I see there are Speedify Custom Bypass entries for a given IP address, but I seem to have to use the policy routes.

TalalMash, 2022-12-09:
> Thanks. Seems to be working now. Of course various streaming services don't like the Speedify VPN (iPlayer / Apple TV / Amazon), so I have had to setup VPN policies to route the smart TV directly to [...]

Great to hear.
Yes I should rename the custom bypass entries `IP addresses` section to `Remote IP addresses`, it's not for LAN IP addresses.

meavydev, 2022-12-10:
The only remaining issue I have is the vpn-policy-routing specifies an interface, but really I want to use Starlink on eth0 and then failover to LTE on eth1 rather than be fixed to one.

Can mwan3 be used with SmoothWAN and vpn-policy-routing? 

I could try duplicating the VPN policy for both AutoUSBEth0 and AutoUSBEth1 and then use mwan3 to failover the interfaces, but will this break Speedify for non-streaming traffic / devices?
 
I am unclear on what process is handling failover for Speedify, or is it handling that itself, as mwan3 doesn't appear to be loaded by default?

TalalMash, 2022-12-11:
Speedify handles everything.
That's a good feature, implementing failover with policy routing, added to project's list for v1.

In the meantime a somewhat tricky solution would be to create 2 SSIDs and 2 additional bridged LAN interfaces with DHCP enabled but with a different limited range by using a smaller subnet.
Then define the two subnets in PBR with the different WANs, most smart devices will hop onto the next saved Wi-Fi network in case the internet goes down so you're relying on per-device failover, which will only work over Wi-Fi.

An easier alternative is to use a load balancer, and plug SmoothWAN to LB's LAN, both using the same subnet for local access on either sides.

On a sidenote, the Speedify's bypass internal tool (found in installation page) works for remote addresses and ports, and it switches the WAN interfaces for the bypass based on data cap, modes and availability. I'm not sure if it's trivial to filter iPlayer and few other services IPs/DNS manually, we can contact Speedify to add these to benefit from those features.
We can also suggest Speedify to add ASN bypass to filter out a range of networks per provider.
Else we may have to find a way to read Speedify's currently selected bypass interface and use it.

And yes mixing mwan3 + PBR is a mess and will need few modifications for using a dedicated routing table and IP rules.

meavydev, 2022-12-11:
Thanks for the detailed response. 

The LG TV is currently on ethernet, so I set it to a VPN bypass as I didn't want to do all the other UK streaming services (ITVX, 4oD, 5...), but we do use iPlayer on other devices, so I added iPlayer using a post from somewhere else which seems to work:

bbctvapps.co.uk bbc.co.uk bbc.com bbc.net.uk sitestat.com bbci.co.uk bbcfmt.hs.llnwd.net bbci.co.uk.edgekey.net loris.llnwd.net bbci.co.uk.edgekey.net

I am actually an Android developer with .Net as well, so I can try helping with development of this feature, although I have yet to set up a working build environment for the Flint OpenWRT. I use a Macbook Pro, so as they seem to favour Linux, I have created an Ubuntu VM and followed their setup instructions. I just haven't tried building yet.
At least I know that the Uboot works to unbrick the Flint, as I have been there already...

meavydev, 2022-12-11:
Well I managed to get a build on Ubuntu 18.04  following the AX1800 instructions here: 
https://github.com/gl-inet/gl-infra-builder#example

[image]

Now I just need to work out what to do to build SmoothWAN.

TalalMash, 2022-12-11:
> Well I managed to get a build on Ubuntu 18.04 following the AX1800 instructions here: https://github.com/gl-inet/gl-infra-builder#example
> 
> [image]
> 
> Now I just need to work out what to do to build SmoothWAN.

I'll post the instructions soon [here](https://smoothwan.com/devl/), a hint: the imagebuilder customization folders also applies to source based builds. 
Hint: You can take a look at the SlateAX folder in SmoothWAN's `devconfig` folder, these are copied to root directory of the OpenWRT subfolder inside the glinet build system.

meavydev, 2022-12-13:
I am still having "fun" trying to get a SmoothWAN build, as copying in the SlateAX devconfig starts giving build breaks in the linux  section due to -Wall and snprintf fails, so I am trying editing the Makefile and changing the -Wall lines to -Wno-errors.
[image]

TalalMash, 2022-12-13:
> I am still having "fun" trying to get a SmoothWAN build, as copying in the SlateAX devconfig starts giving build breaks in the linux section due to -Wall and snprintf fails, so I am trying editing the Makefile and changing the -Wall lines to -Wno-errors. [image]

Lots have changed with the latest Gl.iNet build environment, its easier now.
You need Ubuntu 22.04 upgraded to latest packages.
Added instructions in Wiki and files in SmoothWAN repo.
You may have to remove `rtl8812au-ct` in menuconfig with the latest build.
Added TEST#2 image with better interface naming in Downloads. (untested)

meavydev, 2022-12-13:
Thanks for doing this.
I switched to 22.04 and I think it's fully updated, but of course it doesn't work. I have to admit I always think building Linux is like DLL hell on steroids...
`rtl8812au-ct-2020-01-12-e0d586aa/os_dep/linux/ioctl_cfg80211.c:83:12: error: 'IEEE80211_BAND_2GHZ' undeclared here (not in a function)`
Which sounds like https://bugs.launchpad.net/ubuntu/+source/bcmwl/+bug/1595765

Obviously I got something wrong, but I did start with a new VM image.
I assume it builds for you without any issues?

TalalMash, 2022-12-14:
I had to disable the Realtek module as mentioned in my edit which you may have missed.
The instructions were written based on a test I made yesterday, no issues when building.

meavydev, 2022-12-14:
Ah yes. I had missed the need to go into menuconfig.
Turning off rtl8812au-ct has worked.

meavydev, 2022-12-14:
The .img file was also recognised on a new Flint router (as I didn't want to fiddle with my working one), although it was probably running an older build, as I didn't try updating it at all before applying the firmware img

TalalMash, 2022-12-14:
Great! Do the ports match?
[image]

meavydev, 2022-12-14:
I've disabled ports two and three, as I wanted more LAN ports, but the WAN and One seem to be fine.

Ah.  I just realised that for my .img, the Speedify (and Tailscale) VPN menu entries are missing.

Is this some menuconfig entry I have got wrong?
Everything else seems to be working fine

meavydev, 2022-12-14:
I've been looking at the ports in the Network/Switch section.
It always puts up a dialog warning about topology for Switch0 and the port numbers there are a bit weird.
Not sure if this is what you are after, but:
Port 2 = WAN
Port 3 = LAN 1
Port 4 = LAN 2
Port 5 = LAN 3
Port 1 = LAN 4
This was by plugging in a single ethernet cable  and looking at the disconnected message changing

TalalMash, 2022-12-15:
Ok! I'll change the order soon in `/etc/uci-defaults/99-custom`.
The topology warning can be ignored when NSS offloading is enabled.

I'll recheck the `.config` for the `speedifyconf` and `luci-app-speedifyconf` in menuconfig, is it enabled in your build?

meavydev, 2022-12-15:
I think the problem is the Flint .config doesn't have any mention of Speedify, but the SlateAX does.

TalalMash, 2022-12-15:
> I think the problem is the Flint .config doesn't have any mention of Speedify, but the SlateAX does.

My bad, I've mixed up with the imagebuilder format, added feed update/install commands in Wiki, and updated the current repo `feeds.conf` to relative path.
Uploaded third test image as well.

meavydev, 2022-12-16:
As far as the ports go, I think the ethX numbers are correct, as eth0 is WAN and eth1 is LAN1, which is why I was surprised that the Switch0 page seems to have a different numbering. Almost like someone at GL.iNet got it wrong ;)

meavydev, 2022-12-17:
One comment about the Dev build instructions. I think you need to insert a section after the smoothwan-feeds line saying you need to copy the .config and feeds.conf from the correct directory for your device, so the Flint in my case. 
Otherwise you end up with a stock Flint build, at least I didn't do this and the img was only 12MB with no signs of Speedify and failing update checks for the .img.

I have had to switch my main router to stock OpenWRT MWAN3 now, as my Starlink dish is in a temporary location due to the installer failing the first time (actually one installer never turned up) and they are not coming back until after Christmas. So it's sitting on the ground with plenty of interruptions... Hence my need for MWAN3 failover on the non-Speedify bypass rules.

I am actually finding Speedify to be a bit of a pain, as I have all the pass-throughs turned on, but I have ended up having to add quite a few bypass rules for TV / Chromebook / Tesla. It's not just UK streaming services, but even some websites which were failing. 
The Tesla bypass in Speedify didn't work, as the car said connect to WiFi to download an update when it was...
I can still do some tests for you with my backup router, but at the moment OpenWRT MWAN3 seems to be the best solution for me.

TalalMash, 2022-12-17:
Added the missing step.
Agreed, this is a common issue with most VPNs, the best workaround is to ask for a dedicated residential IP address + dedicated server but it gets expensive. Not to mention the paperwork needed, had few issues with Milkywan residential IP + OMR few years ago.

meavydev, 2022-12-17:
OK thanks.
I followed the updated instructions and it's given me a working .img, although admittedly I had to use from the UBoot menu as it had bricked itself again. A good thing it's the backup...

TalalMash, 2022-12-17:
> OK thanks. I followed the updated instructions and it's given me a working .img, although admittedly I had to use from the UBoot menu as it had bricked itself again. A good thing it's the backup...

Added this issue to TODO list, I'm not sure why it's different than the Slate AX, probably different partition layout or bootloader method.
Thanks

meavydev, 2022-12-18:
I think it's more down to user error on my part.
If the Flint flash update UI puts up a warning about missing manifest / check failure, then you have a bad .img, so don't try flashing via a force (or in this case I tried the TAR).
When the .img passes the check then it is OK, as the .img produced when I did the copy from the flint directory is fine.

TalalMash, 2023-01-14:
Hey @meavydev I'm releasing a new version soon, can you do a ` find /sys/devices/platform/soc/*/net/ | grep eth` and post it here for proper naming?

meavydev, 2023-01-14:
This is on stock OS, not SmoothWAN.
[GLiNet1800_eth.txt](https://github.com/SmoothWAN/SmoothWAN/files/10418399/GLiNet1800_eth.txt)

TalalMash, 2023-01-14:
> This is on stock OS, not SmoothWAN. [GLiNet1800_eth.txt](https://github.com/SmoothWAN/SmoothWAN/files/10418399/GLiNet1800_eth.txt)

That was quick! thanks!
If you want to do a PR on Github as flint contributor replace in `devconfigs/flint/files/etc/hotplug.d/net/00-rename`:
```
#!/bin/sh

if [ $(ls /sys/devices/platform/soc/3a001000.dp1/net/) == "eth0" ]; then
        ip link set eth0 down
        ip link set eth0 name PortWAN
        ip link set PortWAN up
fi
if [ $(ls /sys/devices/platform/soc/3a001200.dp2/net/) == "eth1" ]; then
        ip link set eth1 down
        ip link set eth1 name PortOne
        ip link set PortOne up
fi
if [ $(ls /sys/devices/platform/soc/3a001400.dp3/net/) == "eth2" ]; then
        ip link set eth2 down
        ip link set eth2 name PortTwo
        ip link set PortTwo up
fi
if [ $(ls /sys/devices/platform/soc/3a001600.dp4/net/) == "eth3" ]; then
        ip link set eth3 down
        ip link set eth3 name PortThree
        ip link set PortThree up
fi
```

chansearrington, 2023-02-09:
@TalalMash I've got my Flint arriving this afternoon. Are you dropping a new release soon? The latest release I see says August 2022. 

Is there anything I need to help test?

TalalMash, 2023-02-09:
> @TalalMash I've got my Flint arriving this afternoon. Are you dropping a new release soon? The latest release I see says August 2022.
> 
> Is there anything I need to help test?

Almost, the progress is here (priority): https://github.com/SmoothWAN/SmoothWAN/projects/1
I may send a test build here soon if needed, before full release as there are many additions.

chansearrington, 2023-02-11:
@TalalMash Got my Flint and just installed one of the releases this evening.

[image]

Just a heads up... I can't get Test 3 to work. If I try to upload the .tar file, I get an error. If I extra the tar file, I don't get a .img file, I just get folders with some files in them.

So long story short... I CAN / DID install this one... 

[SmoothWAN-0.99.8HF1-TEST-GLiNet-Flint-WebUI.img]

(https://github.com/SmoothWAN/SmoothWAN/releases/download/0.99.8-flint-test%232/SmoothWAN-0.99.8HF1-TEST-

But I couldn't get this one to give me the .img file after uncompressing the tar.

[SmoothWAN-0.99.8HF1-TEST.3-GLiNet-Flint-WebUI.img.tar]

(https://github.com/SmoothWAN/SmoothWAN/releases/download/0.99.8-flint-test%232/SmoothWAN-0.99.8HF1-TEST.3-GLiNet-Flint-WebUI.img.tar)

meavydev, 2023-02-11:
Sorry if @TalalMash asked me to do a PR, but I have had virtually no internet for a month due to my Starlink router and ethernet adapter failing followed by a faulty long cable...Starlink support taking two weeks to reply... then having to get the installer back to swap to the short cable that Starlink sent to replace the long one, so now we have a cable across the landing floor.
4G was running at 0.4Mbps down with even worse uploads.
 A tail of woe :) 
I just got a new (2021) MacBook Pro, so I needed to setup ARM Ubuntu, but I think I can finally build again.
Let me know if there is anything I can do to help.

TalalMash, 2023-02-12:
@chansearrington @meavydev thanks!
Regarding the TAR file, did you have the same issue when flashing from LuCI?

TalalMash, 2023-02-13:
Latest release should have the updated image format from GL.iNet.
I'm not quite sure of why the size is different yet compared to the Slate AX, added "untested" warning in releases page until a report is posted here.

chansearrington, 2023-02-13:
@TalalMash I only tried to update from the default OS that comes pre-installed on the Flint

TalalMash, 2023-02-13:
> @TalalMash I only tried to update from the default OS that comes pre-installed on the Flint

Added GL.iNet alternative instructions in the new release 0.99.9

zbcochran, 2023-02-17:
Anyone else notice that the Speedify option doesn't show up under "VPN" in 0.99.9?  Am I missing something, or is there another way to install it?

zbcochran, 2023-02-17:
@TalalMash Thank you! Happy to test for you whenever you get it done!

TalalMash, 2023-02-17:
> @TalalMash Thank you! Happy to test for you whenever you get it done!

Uploaded a new build.

zbcochran, 2023-02-17:
Working great so far, thank you for doing this so quickly!

TalalMash, 2023-02-17:
Great to hear, thx for the test.

TalalMash, 2023-02-27:
HF2 released, requires testing before closing this issue as basic support.

meassick, 2023-03-01:
I am now tying to set up a Flint after the Slate but with only 2 WAN ports and three LAN ports.  Previously, it was indicated that "I'm adding a quick switch UI in the next release, in the meantime you'd have to go to Network -> Interface -> Devices (tab) and click on Configure next to br-lan , you can then add or remove bridge ports for LAN access. They're usually eth0, eth1 and so.. I think eth0 is the WAN port and eth4 is the first LAN port, I do not have the device so I'm not sure.
You will also have to remove the extra WANs in the Interfaces tab by clicking Delete next to the network."
Was this quick switch UI implemented, if so , where?

TalalMash, 2023-03-01:
@meassick 
Both Flint and Slate AX need that UI, but a recent change to interface naming requires more testing, especially with USB adapters, once the adapters order is predictable I'll add the UI.
I'll add manual info in the Wiki per devices sometime.

Some hints: delete the desired WANs in `Interfaces`, then go to `Devices` sub-tab and add eth0... (the ones you deleted) to the `Configure` section of `br-lan` in `Bridge ports` list.

closed 2023-12-25 by TalalMash