SmoothWAN/SmoothWAN #77: IP can leak with default setup

DalSoft, 2022-11-04:
I think it would be a good idea to add to the documentation for users to do a [dnsleaktest](www.dnsleaktest.com). This is because people will plug in various ISP configed / locked down devices. For example I plugged in a router / modem from British Telecom that is locked down so I can't change the dns settings or put it in bridge mode. My dns leaked to the ISP dns server 😱

It's a simple fix by changing the "Use DNS servers advertised by peer setting" on the WAN interfaces.

TalalMash, 2023-01-09:
Using DNS forwarding next release, for older version users example:
[image]
[image]

Next version will forward to inactive ADGhome by default.

TalalMash, 2023-02-13:
Fixed in latest release. (no resolvconf)

closed 2023-02-13 by TalalMash