SmoothWAN/SmoothWAN #80: Manually uninstall speedify

neurocis, 2022-11-24:
HOWTO request for removal of Speedify back to the initial fresh install stage. I am finding with interface adjustments the UI gets lost and locked into the "Restart Speedify Service" state.

TalalMash, 2022-11-24:
If you need help customizing the image you can discuss it in discord or github discussion, to remove the files installed, they're in `/usr/share/speedify` and `speedifyui` folder, here is the script responsible for installation: https://github.com/TalalMash/smoothwan-feeds/blob/main/speedifyconf/net/usr/lib/speedifyconf/run.sh

neurocis, 2022-11-24:
Thanks, I have tried removing those and there must be a config remnant as the UI still does not work after re-install. Even tunneling ports 9330 and 9331 via SSH so they are on my localhost gives "Restart Speedify Service" error.

TalalMash, 2022-11-24:
Speedify WebSocket service listens to localhost only, there is a NGINX relay to proxy requests outside localhost, check the configuration at `/etc/nginix` if there is a need for modifications.

neurocis, 2022-11-25:
Understood re localhost only websocket, one can work around via SSH port forward. Anyhow non-consequential to my issue. For some reason I am unable to get speedify back to a sane aka "configurable via gui" state after changing br-lan to have two IP's.

neurocis, 2022-11-25:
Poking around I did find this which seems odd to me:

```
root@SmoothWAN:/usr/share/speedifyui# ./speedify_ui
./speedify_ui: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```

TalalMash, 2022-11-25:
Speedify UI is a web app and that is the executable for the desktop viewer, the UI is served with NGiNX.

For two IPs, check the custom firewall entries, manually specify the IPs there for port/addr rewrite.

neurocis, 2022-11-25:
Bing, that has got it ... uci returns format iptables is not familiar with ...
```
root@SmoothWAN:~# uci get network.lan.ipaddr
192.168.4.1/24 172.67.1.1/24
```

changed custom rule:
`iptables -t nat -A PREROUTING -d $(uci get network.lan.ipaddr)/32 -p tcp -m tcp --dport 9330 -j DNAT --to-destination 127.0.0.1:9331`

to be:
`iptables -t nat -A PREROUTING -i br-lan -p tcp -m tcp --dport 9330 -j DNAT --to-destination 127.0.0.1:9331`

That casts a wider net (potential false positives, may be edge) so not submitting a PR, leave it to your discretion. Thanks for your assistance!

Cheers!

TalalMash, 2022-11-25:
You can duplicate the entry with the other IP, else services using this port on the internet will be redirected to the router, breaking the connection. And accessing the Speedify UI from Tailscale may not work with interface filtering.

neurocis, 2022-11-25:
Agreed, I did that initially but I wanted a more automated solution and the false positives on that port I do not see affecting me. I may try and script it proper-like, if I do i will submit a PR.

TalalMash, 2022-11-28:
Will read UCI entry issue in firewall in JSON and parse accordingly in the next version for multi-IP LAN, changing title.

TalalMash, 2023-01-07:
Note: `&wsPort=9331` in Speedify UI URL parameter will replace firewall next release.

closed 2023-01-07 by TalalMash