SmoothWAN/SmoothWAN #87: Constantly getting RRDtool Plugin Errors if I have ports forwarded

chansearrington, 2023-01-04:
Hardware is a Gl.inet Slate
Running v0.99.8
With two incoming WANs
- Broadband Cable Modem
- T-Mobile 5G Router

I'm currently using VPN PBR to my website server (and other things) set to only use "PortONE" (the cable WAN) because it supports port forwarding and the T-mobile one doesn't (or so I've been told)

I also have Port Forwarding enable for the specific ports that I need open / routed (images of both configs below)

I'm not sure if this is a U.S.E.R. error or a bug (probably me :-)) Or if I'm even doing this correctly or not. 

"Cronos" (192.168.1.142) is my website server

[image]

[image]

chansearrington, 2023-01-04:
I'm not sure if this is related or not, but after some period of time, I'll notice my internet speed has dropped and I'll look to find out that Speedify is no longer running. Clicking the "restart speedify" button doesn't do anything. But after a reboot everything is back to normal.

Not sure if this is related or not.

TalalMash, 2023-01-06:
Gotta replicate that before the next release, next version will include a
menu to download all logs while removing passes.

TalalMash, 2023-01-07:
I'm getting the same harmless errors after reboot but with single instance of rrd_update_r per file which is normal since on restart/power-cycling there is a delay in syncing UBIFS.
It's normal to have these errors while changing firewall settings or on startup since OpenWRT is a collection of software that are not related to each other (no ecosystem to manage).
I couldn't find functionality issue, regarding the last comment feel free to open a new issue with logs generated using the `Generate Logs` button in SmoothWAN's Speedify menu. (navbar)

TalalMash, 2023-01-07:
Note to self: Modify collected to to wait for NTP/dnsmasq service.

TalalMash, 2023-01-09:
Removed my previous comment to avoid confusion:
Apparently the Slate/Flint UBIFS in uboot appears to be very old <4.14 kernel, please remove this and report back if the errors are frequent:
[image]

chansearrington, 2023-01-10:
@TalalMash Done! I'll let you know how it goes.

chansearrington, 2023-01-11:
@TalalMash Well... I don't understand what's happening in this log, lol. I'm including the latest System Log and Kernal logs below.

I'm not sure if this is the same problem or a new problem, but for some reason the Speedify service stops running after a certain amount of time. It's done this since day 1 on the Slate. After I rebooted to remove the line you mentioned above, this time it made it a little over 14 hours and stopped around at 00:40:27 local time

After the service stops, it shows a button of "Restart the Speedify service". (see pic)

[image]

Clicking this button doesn't seem to do anything. The service never restarts.

The only way to get the Speedify service restarted (that I know of) is to reboot the device.

After the restart, the service automatically launches and all if right in the world again. :-)

[System Logs](https://pastebin.com/1vNy807V)

[Kernel Logs](https://pastebin.com/fa7aS1YR)

chansearrington, 2023-01-11:
@TalalMash this time it lasted 12 hours.  (logs and image below)

[image]

[System Logs](https://pastebin.com/31zVhSR6)

[Kernel Logs](https://pastebin.com/rZbEU9Bh)

TalalMash, 2023-01-11:
> @TalalMash <https://github.com/TalalMash> you can tell how long before
> the error by the "Uptime" time counter row on the Overview page screenshots
> above.

It looks like few of the Speedify log files are corrupt, and unfortunately
Speedify reads logs as resume data, does it happen again when you reset
Speedify from the setup menu?
The corruption is probable from the now deleted mount option.

TalalMash, 2023-01-12:
I'm releasing a new version in the next few days if you need to start over, the kernel log shows Speedify with a memory leak. You can also start over with the current version with that mount line removed, thanks for the report.

chansearrington, 2023-01-12:
@TalalMash Yeah, looks like memory leak in the graph as well.

14:00 was my last reboot of the device and there's a steady memory increase until about 21:00 (this is when I reset Speedify through the Speedify area in the menu.

Then another steady increase until it crashed at about 05:30

[image]

chansearrington, 2023-01-12:
@TalalMash Looking at the Processes tab, the only thing that's steadily increasing is ./speedify -d logs

It started under around 24% after reboot and is slowly / steadily increasing. Currently sitting at 42%

[image]

chansearrington, 2023-01-12:
@TalalMash 3.75 hours in and it's at 53%

TalalMash, 2023-01-12:
> [image: CleanShot 2023-01-12 at 08 56 ***@***.***

That's interesting, this is highly likely to be a Speedify bug. Any weird
names or characters with the WAN interfaces? What settings are used And how
many WANs are there.
You can submit the logs using the generate logs to Speedify for further
info that may help, I'll try to replicate it based on your configuration.

TalalMash, 2023-01-13:
Continue in https://github.com/SmoothWAN/SmoothWAN/issues/89

TalalMash, 2023-02-13:
Fixed in latest release.

closed 2023-02-13 by TalalMash