SmoothWAN/SmoothWAN #94: Support for NETGEAR R7800 (Nighthawk X4S AC2600)

TalalMash, 2023-01-24:
**Status:** In development
Released in v0.99.9

Need testers.

benbeckner, 2023-01-24:
Good morning!
Can you provide details on what changes are in this version and what you need tested?

Thanks!

TalalMash, 2023-01-24:
@benbeckner lots of changes! releasing the image soon with the changelogs.
Port order and configuration, will post the instructions here, thanks.
Supporting Netgear over Belkin since they're difficult to brick (even intentionally) and no restrictions to 3rd party firmware.

TalalMash, 2023-01-24:
Unfortunately these appear to be mostly out of stock and chip is old and overpriced in the new lineup, sorry for the noise! Replacing with the recent WAX series which will be pushed to official OpenWRT soon!

closed 2023-01-24 by TalalMash