SmoothWAN/SmoothWAN #95: (Bonding) Device support suggestions thread

TalalMash, 2023-01-24:
**UX**
- Must be able to flash official OpenWRT from the OEM firmware page with minimal effort.
- TFTP or other recovery methods available, locked bootloader partitioning is preferable (common with stock UBIFS).
- AES-crypto hardware acceleration or single thread high performance CPU (>RPi4).
- Mainline (snapshot) or current release official OpenWRT support.
- Proprietary OpenWRT forks like Qualcomm's semi-proprietary QDSK (e.g GL.iNet) or Rockchip BSP with open source build-system is supported.
- Due to the poorly performing modern Wi-Fi USB sticks on Linux, SoC Wi-Fi or PCI-E Wi-Fi is required.
- SDIO Wi-Fi is not supported (found in most single board computers)
- In production or discontinued hardware up to 2 years.

**Hardware specs**
- ARM/x86 CPUs only (ARM32/64-bit and x86 64-bit)
- ARM minimum requirement: 700+Mhz dual for Cortex-A53 or higher
- Memory: DDR3L 256Mbyte+
- Storage 128Mbyte+
- Reverse engineered hardware support must be in OpenWRT mainline with caveats listed, examples:
  - IPQ80xx series
  - MT76xx series
  - Updated modern hardware list: https://openwrt.org/toh/views/toh_available_16128_ax-wifi 
  (AX hardware usually meets minimums)
- 3 or more ethernet ports
- 1 or more USB ports

TalalMash, 2023-01-24:
My current candidates (may be added):
Netgear WAX202: Budget, a bit slow CPU, 800Mhz but with AES and network HW offloading, no issues with Apple AX Wi-Fi hardware (common with Mediatek OpenWRT)
Netgear WAX206: Much faster than the WAX202, still in snapshot.
Belkin RT3200: Apple Wi-Fi degradation only when AX Wi-Fi is enabled

TalalMash, 2023-01-26:
Banana Pi R64 and R3 will be added (in production)

TalalMash, 2023-02-21:
[GL.iNet Spitz AX](https://www.gl-inet.com/products/gl-x3000/#specs), may have to keep GL.iNet portal page for 4G/5G modem support.

cabinlab, 2023-03-12:
> [GL.iNet Spitz AX](https://www.gl-inet.com/products/gl-x3000/#specs), may have to keep GL.iNet portal page for 4G/5G modem support.

Are you still considering adding Spitz support? An option with built-in dual SIM would be pretty sweet!

bt61, 2023-03-12:
> > [GL.iNet Spitz AX](https://www.gl-inet.com/products/gl-x3000/#specs), may have to keep GL.iNet portal page for 4G/5G modem support.
> 
> Are you still considering adding Spitz support? An option with built-in dual SIM would be pretty sweet!

The Speedify team has pointed out on their livestream that phones with dual-sim only have one actual modem and simply switch quickly between the sims to make both somewhat active. As a result, they say, it's not useful for Speedify.  It seems like it could still be advantageous in some cases, but I'll believe them that it's not generally a good idea.

I looked at the Spitz page and nothing indicates it actually has 2 modems, so I think it's safe to assume for now that it's a single modem with dual sim, so maybe not useless for Speedify, but not as ideal as initially it may appear.

TalalMash, 2023-03-12:
@bt61 that's correct and it's single modem.
@cabinlab Yes once it's released, signed up for beta testing.

kputland, 2023-05-03:
NanoPi R6S

TalalMash, 2023-05-08:
> is it necessary to have Memory: DDR3L 512Mbyte+ ?

Yes, Speedify 12+ seem to use more memory now.

TalalMash, 2024-01-08:
Spitz AX and Beryl AX can try https://github.com/TalalMash/Unofficial-Speedify-Installer-For-OpenWrt
Beryl AX is reported to be working.