SmoothWAN/SmoothWAN #99: GPIO on Raspberry Pi 4

BigWillOKC, 2023-02-06:
Hello, I'm coming over from OpenMPTCProuter and would like to enable the same functionality I have on my Pi with that setup.  I have a momentary switch connected to my Pi on GPIO 3 (pins 5 and 6) that allows me to shutdown and restart the Pi with a button press.  When running SmoothWAN, it will boot the Pi if its already shutdown, but it will not execute the shutdown when I press the button.  Is there something I need to do differently?  I'm using the following entry in my config.txt

# Power Button Shutdown/Startup
dtoverlay=gpio-shutdown
# Power Button Status LED
gpio=15=op,dh

TalalMash, 2023-02-06:
Hello!
Don't have the Pi anymore (hit by ⚡), the `kmod-gpio-button-hotplug` is missing by default, is that installed?

BigWillOKC, 2023-02-06:
That was it, thank you so much!  Any idea when PnP mini display monitor support will be added?

TalalMash, 2023-02-06:
Great will add it next release, regarding the display: it's already supported in version 0.99.8, but the choice of display will change in v1 due to the SSD1306 having short lifespan.

TalalMash, 2023-02-06:
Speedify UI fullscreen link is unauthenticated, you can use a spare (old) phone as an alternative.
M5 stack may be the default in v1 with USB & MQTT-over-network support.

BigWillOKC, 2023-02-06:
That's a good idea.  I'm building a mobile internet system in a small Pelican style case, so i would love to be able to see that speedify ui really easily or at least the stats.  It would be ideal if i could have it show on the 5" screen i have installed for the Pi, but guessing that's a no go.

TalalMash, 2023-02-06:
Adding a TUI is possible for plug and play devices, if the 5 inch display is using HDMI and OpenWRT console is visible then it's feasible to implement, this will also benefit x86/mini-pc, this is definitly in plan.

TalalMash, 2023-02-13:
Opening a new issue regarding the display on any update.

closed 2023-02-13 by TalalMash