SmoothWAN/SmoothWAN #105: Torrent or P2P Traffic Detected

ccmks, 2023-02-16:
Hello,

I am not sure if you can help on this. My speedify keep disconnecting because it says it is detected torrent or p2p traffic. However, I am not using those at all. How speedify identify my traffic as torrent or p2p traffic so I can filter those packet out and not to use speedify network?

Thank you

TalalMash, 2023-02-16:
In the latest release, I added this https://smoothwan.com/tips/#identifying-client-using-p2p-or-bittorrent to identify the clients using BitTorrent.
Speedify is tame with Windows P2P and other non torrenting traffic.

I also added "ntopng" which is a heavy tool, but is able to identify the type of traffic in very high details using DPI

ccmks, 2023-02-16:
I was using that tool earlier but didn't show anything on BitTorrent activity log, but speedify still flag my connection as P2P

TalalMash, 2023-02-16:
It's probably not torrent, I'm not aware of what Speedify considers P2P, I wish there was a list in support page.
You can try using `ntopng` in services menu, and select `br-lan`, then `Live flows` in the sidebar.

ccmks, 2023-02-17:
I contact speedify support and they want me to generate log. How can I generate the log?

I tried to click generate log in UI but nothing happens

TalalMash, 2023-02-17:
In SmoothWAN Speedify menu (where you install) there is a generate log button.

TalalMash, 2023-02-27:
Speedify related, move to discussions.

closed 2023-02-27 by TalalMash