SmoothWAN/SmoothWAN #118: Error when updating SmoothWAN from SmoothWAN

chansearrington, 2023-02-26:
I have firmware SmoothWAN 0.99.8 Stable on my GL-iNet Flint router 
 and I finally had time today to upgrade to the 99.9 Beta WebUI release for Flint and tried to upgrade SmoothWAN from Admin / System / Flash
 
 I got the following error message.
 
Image check failed:

Sat Feb 25 23:56:37 UTC 2023 upgrade: Device glinet,axt1800 not supported by this image Sat Feb 25 23:56:37 UTC 2023 upgrade: Supported devices: glinet,ax1800 Image check failed. 
 
[image]

I forced the update and made sure that the settings were not maintained (as instructed in the release notes) 

Everything seems fine, lol. But I'm not smart enough to know if there's a problem anyway.

TalalMash, 2023-02-26:
It is ok as it's not the standard OpenWrt format.

closed 2023-02-26 by TalalMash

TalalMash, 2023-02-27:
Linked to release page as a note.