SmoothWAN/SmoothWAN #122: Custom USB/device assignments on Pi 4/feeding links through USB hub

cr08, 2023-03-02:
Not sure how possible this is. I'm fairly familiar with OpenWRT and mucking with interfaces but not sure how this interacts with SmoothWAN.

I have a Pi 4 running currently and have a few phones on different carries tethered via USB. However I'm finding I'm having some power issues. I have a proper powered hub with USB BC so I can reliably provide up to 1.5a per device. But because it is feeding into the Pi as a hub and one USB port, SmoothWAN doesn't know what to do with it.

I tried researching the repo here and the website but didn't see anything specific on this topic.

TalalMash, 2023-03-02:
Power the Pi, wait till it finishes booting, plug in the hub, wait few seconds, then post the log output of the Kernel from `Status` -> Logs/Kernel section.

cr08, 2023-03-03:
Huh. So I reloaded SmoothWAN from scratch to make sure I started from a clean slate as I had tried messing with the interfaces previously to work around this (mainly just manually assigning the usbX interfaces which seemed to work but meant I lost any automated assignments in the process) and it seems to work fine now. Not sure what caused it to misbehave previously. But here's the kernel log from the clean run anyways even though it's probably moot at this stage.

I guess my original concern after initial testing was that it was very specific about tracking the individual ports on the Pi with no facilities for hubs but looks like that's not the case.

````
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.161 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20028-43d71ad93e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Jan 3 00:24:21 2023
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45848 r8192 d31976 u86016
[    0.000000] pcpu-alloc: s45848 r8192 d31976 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0B:3F:32 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 873544K/970752K available (7934K kernel code, 902K rwdata, 1284K rodata, 448K init, 545K bss, 31672K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000229] Console: colour dummy device 80x25
[    0.000502] printk: console [tty1] enabled
[    0.000534] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000560] pid_max: default: 32768 minimum: 301
[    0.000704] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000739] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001375] cgroup: Disabling memory control group subsystem
[    0.002382] rcu: Hierarchical SRCU implementation.
[    0.002566] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003107] smp: Bringing up secondary CPUs ...
[    0.003552] Detected PIPT I-cache on CPU1
[    0.003613] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.004126] Detected PIPT I-cache on CPU2
[    0.004167] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004666] Detected PIPT I-cache on CPU3
[    0.004706] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004782] smp: Brought up 1 node, 4 CPUs
[    0.004839] SMP: Total of 4 processors activated.
[    0.004853] CPU features: detected: 32-bit EL0 Support
[    0.004868] CPU features: detected: CRC32 instructions
[    0.004941] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004959] CPU: All CPU(s) started at EL2
[    0.004989] alternatives: patching kernel code
[    0.013001] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013051] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014734] pinctrl core: initialized pinctrl subsystem
[... 591 lines cut ...]
[  637.231558] usb 1-1.2.4: new high-speed USB device number 9 using xhci_hcd
[... cut ...]
[  645.951687] usb 1-1.2.4: new high-speed USB device number 11 using xhci_hcd
[  646.083841] usb 1-1.2.4: New USB device found, idVendor=05c6, idProduct=9024, bcdDevice= 3.10
[  646.092516] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  646.100022] usb 1-1.2.4: Product: K88
[  646.103792] usb 1-1.2.4: Manufacturer: ZTE
[  646.107903] usb 1-1.2.4: SerialNumber: 58f8cc5f
[  646.117760] rndis_host 1-1.2.4:1.0 usb2: register 'rndis_host' at usb-0000:01:00.0-1.2.4, RNDIS device, 02:56:0d:00:38:63
````

TalalMash, 2023-03-03:
thats great! Did you by chance use the HF2 version previously? HF3 is port agnostic, no more interface renaming.

cr08, 2023-03-05:
I just had a chance to check the image file on the machine I originally grabbed it on and it looks like I started with HF1 so that probably a lot. So far so good with HF3 though at this point. :)

TalalMash, 2023-03-06:
Better interface naming and hub support in `0.99.9HF4`

closed 2023-03-06 by TalalMash