SmoothWAN/SmoothWAN #123: Speedify installed, but stuck on Restart Speedify button.

Sealilly76, 2023-03-03:
Hey all, installing on Slate, all went smooth until I installed Speedify. Log says installed succesfully but at Status:Overview, I only have the blue Restart Speedify button. No joy after clicking. Uninstalled successfully, reinstalled, rebooted, still no joy. Any ideas?

Sealilly76, 2023-03-03:
Seemed to be a Firefox issue, worked fine with Chrome.

TalalMash, 2023-03-03:
Thanks for the report, were there any extensions e.g adblock?

TalalMash, 2023-03-03:
Same issue here with FF, happens with other WebUI too like Ubiquity, add exception to websocket:
[image]

TalalMash, 2023-03-03:
Also needed for web terminal with port `7681`

TalalMash, 2023-03-03:
Next version (v1) will have http by default, https still available.

Sealilly76, 2023-03-04:
Thank you, that worked! This setup is AWESOME.

TalalMash, 2023-03-06:
Fixed in `0.99.9HF4` , clear browser history related to `172.17.17.2` and cache. 
Visit with `http` instead of `https`

closed 2023-03-06 by TalalMash