SmoothWAN/SmoothWAN #127: Support for BananaPi R3

TalalMash, 2023-03-06:
**Status**: Basic support image release, needs testers.

kputland, 2023-05-08:
Where do you think this fits performance wise with the Slate AX and the NanoPi R6S? I like the idea of the eight antennas. And price wise it's in the same ballpark if you don't include the 2.5G SFPs.

TalalMash, 2023-05-08:
It should be slightly slower than the R6S.

TalalMash, 2023-05-08:
This got released too: https://www.cnx-software.com/2023/03/14/banana-pi-bpi-r3-mini-low-profile-2-5gbe-router-board-mediatek-filogic-830-soc/

kputland, 2023-05-13:
Should have a RPI-R3 on Monday

kputland, 2023-05-15:
Boots from image `SmoothWAN-0.99.9HF5-BananaPi-R3-SDcard.img.gz` , but it's on 192.168.1.1 and I don't see any customizations. no webUI. SSH fingerprint changes on reboot. Will try a vanila OpenWRT

kputland, 2023-05-15:
Will have to come back to this device in about a month. I won't have much time until mid June.

TalalMash, 2023-05-16:
Seems to be booting from NAND/NOR that is shipped with the device, not SDcard. Or you may have to install using: https://forum.banana-pi.org/t/banana-pi-bpi-r3-openwrt-image/13236/82

kputland, 2023-10-12:
Checked jumpers. changed images.
When I flash the official image
`openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz`
banner repors ` OpenWrt SNAPSHOT, r22884-22ff416483`

When I flash `SmoothWAN-0.99.9HF5-BananaPi-R3-SDcard.img.gz`
banner reports ` OpenWrt SNAPSHOT, r22480-18d516a649`

which leads me to believe that it is booting from the SD card.

There is a very quick menu in the serial console at boot
[image]

The only option that works is `1. Run default boot command` or `4. Boot recovery system from SD card.`

TalalMash, 2023-12-25:
Reopening when I can locally test it.

closed 2023-12-25 by TalalMash