SmoothWAN/SmoothWAN #130: Diagnostics page, download test, only cownloads a couple of bytes

tiernano, 2023-03-09:
Not sure if this is SmoothWAN or an upstream issue, but while trying to use the diagnostics page (/cgi-bin/luci/admin/network/diagnostics) and using the Download Test, it only downloads about 3k and stops. A bit of digging shows the URL (https://releases.ubuntu.com/22.04.1/ubuntu-22.04.1-desktop-amd64.iso) has a 301 on it to https://old-releases.ubuntu.com/releases/22.04.1/ubuntu-22.04.1-desktop-amd64.iso. 

I have searched though the code in SmoothWAN for that URL but can't seem to find it, which makes me think it's upstream.

TalalMash, 2023-03-09:
`curl` command is missing the `-L` flag to follow redirects, fixed in next release, to patch do:
```
wget https://raw.githubusercontent.com/SmoothWAN/SmoothWAN-feeds/main/speedifyconf/net/usr/bin/wanspeedtest -O /usr/bin/wanspeedtest
```
Thanks for the report.

tiernano, 2023-03-09:
just patched the file and ran the test and its working now. Thanks!

TalalMash, 2023-04-03:
Fixed in 0.99.9HF5.

closed 2023-04-03 by TalalMash