SmoothWAN/SmoothWAN #135: v0.99.9 HF4 ngnix errors in logfile

pbix, 2023-03-14:
I am running v0.99.9 HF4 and observe the below error messages in the logfile.  I think these are caused by a WebUI login from my iOS device.  Not sure they are important or not but thought I would report.  I also believe these existed in a previous versions as well.

This is a great project BTW, really appreciate you efforts on it.

Thanks,
pbix

`Tue Mar 14 19:02:16 2023 user.info : luci: accepted login on / for root from 192.168.2.226
Tue Mar 14 19:02:17 2023 daemon.err nginx[8674]: 2023/03/14 19:02:17 [error] 8677#0: *7623 open() "/www/spdui/WorkSans-Regular.ttf" failed (2: No such file or directory), client: 192.168.2.226, server: _lan, request: "GET /spdui/WorkSans-Regular.ttf HTTP/1.1", host: "192.168.2.10", referrer: "http://192.168.2.10/spdui/?uid=969422&wsPort=9331"
Tue Mar 14 19:02:17 2023 daemon.err nginx[8674]: 2023/03/14 19:02:17 [error] 8678#0: *7619 open() "/www/spdui/WorkSans-SemiBold.ttf" failed (2: No such file or directory), client: 192.168.2.226, server: _lan, request: "GET /spdui/WorkSans-SemiBold.ttf HTTP/1.1", host: "192.168.2.10", referrer: "http://192.168.2.10/spdui/?uid=969422&wsPort=9331"
Tue Mar 14 19:02:17 2023 daemon.err nginx[8674]: 2023/03/14 19:02:17 [error] 8677#0: *7621 open() "/www/spdui/WorkSans-Bold.ttf" failed (2: No such file or directory), client: 192.168.2.226, server: _lan, request: "GET /spdui/WorkSans-Bold.ttf HTTP/1.1", host: "192.168.2.10", referrer: "http://192.168.2.10/spdui/?uid=969422&wsPort=9331"
Tue Mar 14 19:02:17 2023 daemon.err nginx[8674]: 2023/03/14 19:02:17 [error] 8678#0: *7620 open() "/www/spdui/WorkSans-Light.ttf" failed (2: No such file or directory), client: 192.168.2.226, server: _lan, request: "GET /spdui/WorkSans-Light.ttf HTTP/1.1", host: "192.168.2.10", referrer: "http://192.168.2.10/spdui/?uid=969422&wsPort=9331"
Tue Mar 14 19:02:17 2023 kern.warn kernel: [43993.435352] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=597 TOS=0x02 PREC=0x00 TTL=64 ID=12871 DF PROTO=TCP SPT=9330 DPT=51942 WINDOW=128 RES=0x00 ACK PSH URGP=0
Tue Mar 14 19:02:17 2023 kern.warn kernel: [43993.877821] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=597 TOS=0x02 PREC=0x00 TTL=64 ID=12884 DF PROTO=TCP SPT=9330 DPT=51942 WINDOW=128 RES=0x00 ACK PSH URGP=0
Tue Mar 14 19:02:18 2023 kern.warn kernel: [43994.169958] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=644 TOS=0x02 PREC=0x00 TTL=64 ID=13140 DF PROTO=TCP SPT=9330 DPT=51942 WINDOW=128 RES=0x00 ACK PSH URGP=0
Tue Mar 14 19:02:33 2023 daemon.n`

TalalMash, 2023-03-15:
Thanks glad it's helpful, the fonts should be moved to `/www/spdui/` to load Speedify defined font when using the browser instead of the app. Will fix next release. They're pretty harmless on the latest browsers.

TalalMash, 2023-04-03:
"torrent" is mentioned in Speedify UI websocket data, nothing much to do with `nftables` input chain.
Seems like the fonts are already loaded, they're needed for the desktop clients, you can safely ignore.

closed 2023-04-03 by TalalMash