SmoothWAN/SmoothWAN #136: v0.99.9 HF4 - FLINT - DHCP ISSUE

chansearrington, 2023-03-15:
Hey @TalalMash,

I've flashed SmoothWAN v0.99.9 HF4 on my GL.iNET Flint device and I think I'm running into an issue (sorry, I'm no expert, but if you tell me what to check I can look into it)

I think (not sure) that the Flint version is missing the DHCP / DNS capabilities.

There are no devices listed in the Active DCHP leases.
[image]

If I go to Network > DCHP and DNS... the screen seems to show that I don't have dnsmasq install.

[image]

If I change the main address network address from 172.17.17.2 to something like 192.168.1.1 then the router doesn't even hand out any kind of DCHP and tries to pass the external modem address directly through... which doesn't work. The only way for me to access the router is to manually define the router / ipaddress / subnet

Here's some other screenshots in case they help too.
[image]

chansearrington, 2023-03-15:
If I revert all the way back to [SmoothWAN-0.99.8HF1-TEST.3-GLiNet-Flint-WebUI.img.tar](https://github.com/SmoothWAN/SmoothWAN/releases/download/0.99.8-flint-test%232/SmoothWAN-0.99.8HF1-TEST.3-GLiNet-Flint-WebUI.img.tar)

then the problem goes away.

But I have to go all the way back to 0.99.8

the problem persists in all the 99.9 releases. I checked them each individually.

bt61, 2023-03-15:
Not sure about Flint, but for RPi all the DNS and DHCP got moved to AdGuard Home, check there (under Services).

TalalMash, 2023-03-15:
Correct, added the notice in the main page too (it was hidden in changelog).
Simply change the IP of LAN and the range/address in AdGuard Home.

[(if you need dnsmasq back for other reasons)](https://smoothwan.com/adg/#due-to-popular-request-and-leak-proof-dns-adgh-is-the-default-dhcp-server)

chansearrington, 2023-03-17:
Got it and that worked. Thanks!

I don't think I needed / wanted dnsmask. there just wasn't anything in the UI telling me where to edit the DHCP range or that dnsmask was replaced with AdGuard.

TalalMash, 2023-04-03:
Reverted to stock openwrt dhcp configuration and dnsmasq but with working adguardhome client statistics.
Fixed in 0.99.5HF5

closed 2023-04-03 by TalalMash