SmoothWAN/SmoothWAN #141: Speedify Stops and cannot be restarted

pbix, 2023-03-22:
I am running v0.99.9 HF4, x86 under proxmox in a VM.  I have two internet links.  After a few days of operation the Speedify service just stops and the only way to restart it is to reboot VM.   It seems in some ways similar to [https://github.com/SmoothWAN/SmoothWAN/issues/89](url).

I have collected some data with the hopes of getting some help here on figuring out what is going on.

First is the screenshot I see.  The restart button has no effect
[image]

Second is the main screen of proxmox which shows the memory levels.  I do not see anything wrong here.
[image]

Third is the kernel log.  Some strange PSP messages there which I do know the meaning of.
[dmeesg.txt](https://github.com/SmoothWAN/SmoothWAN/files/11035081/dmeesg.txt)

Fourth is the system log.  There is a strange nlbwmon "out of memory error" on line 6888 (3/19/23:33:49)
`Sun Mar 19 23:33:49 2023 daemon.err nlbwmon[7037]: Netlink receive failure: Out of memory
Sun Mar 19 23:33:49 2023 daemon.err nlbwmon[7037]: Unable to dump conntrack: No buffer space available
`

Following that I see periodic "Objject busy errors".
`Mon Mar 20 00:47:34 2023 daemon.err nlbwmon[7037]: Netlink receive failure: Object busy
Mon Mar 20 00:47:34 2023 daemon.err nlbwmon[7037]: Unable to dump conntrack: I/O error
`

I cannot say for certain when Speedify when down (I was sleeping) but it was that night so I do think these messages might be related to that.
[logread.txt](https://github.com/SmoothWAN/SmoothWAN/files/11035101/logread.txt)

This is happening every 3 or 4 days here so I will get another chance to collect data if someone can tell me what to look for.

pbix, 2023-03-22:
Some update on this.  After running only 24 hours this time speedify stopped again.
This time I see no nlbwmon errors in the syslog.  But in the kernal log I see the following:

``traps: speedify[12868] general protection fault ip:55ebcc9d71d2 sp:7ffeb93fd830 error:0 in speedify[55ebcc76c000+519000]``

bt61, 2023-03-22:
Did you confirm by checking your public IP that Speedify was down and not just an issue with that UI element?  I have had that happen before, though not recently and on different hardware.

TalalMash, 2023-03-22:
I'll run one in Proxmox for few days, but this seems like a RAM hardware issue or Proxmox qemu configuration, will see, any other customization to replicate here? Try running memtest in Proxmox to be sure.

pbix, 2023-03-25:
I will run the Memtest tomorrow when the wife is out of town but thought that I would report today another crash with the following interesting message in the log.

Fri Mar 24 20:30:34 2023 kern.info kernel: [135544.744678] traps: speedify[12785] general protection fault ip:7f9fe90eff68 sp:7ffd1dc483f0 error:0 in libc-2.31.so[7f9fe908c000+15a000]

pbix, 2023-03-25:
Memtest run today found no memory issues.

@TalalMash My Proxmox settings are below.  Can you compare with yours to see if you see any meaningful difference?  I am running.  I am running SmoothWAN in a VM using ProxMox 7.3-4.

[image]

[image]

TalalMash, 2023-03-25:
[image]
[image]

Ill update `libc` next release, you can downgrade Speedify to ~12.8 or similar to see if the newest update requires newer libraries.

TalalMash, 2023-03-25:
[image]

pbix, 2023-03-25:
I am using SmoothWAN 13.1.0.10834.  What version are you using?

TalalMash, 2023-03-25:
> I do not see any meaningful difference in the Proxmox Configs. I am using SmoothWAN 13.1.0.10834. What version are you using?

Same version.

TalalMash, 2023-03-25:
Upgrade libc to 2.36 test
```
wget -c http://ftp.de.debian.org/debian/pool/main/g/glibc/libc6_2.36-8_amd64.deb
ar x *.deb
tar -xvf data.tar.xz
cp etc lib64 lib usr / -rP
```

pbix, 2023-03-25:
The CPU is 4 x Intel(R) Core(TM) i5-4590T CPU @ 2.00GHz (1 Socket).

I looked at libc6_2.36 package.  Have you done this upgrade on your setup?  I see 2.37 is also released.

TalalMash, 2023-03-25:
No my setup is stock.

pbix, 2023-04-05:
As an update on this topic I updated Proxmox from SMP PVE 5.15.83-1 to SMP PVE 5.15.102-1.  The frequency of occurrence may have reduced but the crash still occurs.

The kernel log on my rig is continuously spammed with the below messages starting about 80 seconds after boot.  Not sure its related but it is annoying.
```312940.734738] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=595 TOS=0x00 PREC=0x00 TTL=64 ID=51710 DF PROTO=TCP SPT=9330 DPT=49482 WINDOW=128 RES=0x00 ACK PSH URGP=0 
[312940.739742] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=595 TOS=0x02 PREC=0x00 TTL=64 ID=51708 DF PROTO=TCP SPT=9330 DPT=49482 WINDOW=128 RES=0x00 ACK PSH URGP=0 
[312940.746394] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=595 TOS=0x02 PREC=0x00 TTL=64 ID=35762 DF PROTO=TCP SPT=9330 DPT=53278 WINDOW=128 RES=0x00 ACK PSH URGP=0 
[312940.758509] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=595 TOS=0x02 PREC=0x00 TTL=64 ID=51714 DF PROTO=TCP SPT=9330 DPT=49482 WINDOW=128 RES=0x00 ACK PSH URGP=0 
[312940.765573] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=595 TOS=0x02 PREC=0x00 TTL=64 ID=51942 DF PROTO=TCP SPT=9330 DPT=34316 WINDOW=2302 RES=0x00 ACK PSH URGP=0 
[312941.434098] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=663 TOS=0x02 PREC=0x00 TTL=64 ID=36041 DF PROTO=TCP SPT=9330 DPT=53278 WINDOW=128 RES=0x00 ACK PSH URGP=0 
[312941.439938] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=663 TOS=0x02 PREC=0x00 TTL=64 ID=52221 DF PROTO=TCP SPT=9330 DPT=34316 WINDOW=2302 RES=0x00 ACK PSH URGP=0 
[312941.447078] P2P:IN=lo OUT= MAC=00:00:00:00:00:00:00:00:00:00:00:00:08:00 SRC=127.0.0.1 DST=127.0.0.1 LEN=663 TOS=0x02 PREC=0x00 TTL=64 ID=51998 DF PROTO=TCP SPT=9330 DPT=49482 WINDOW=128 RES=0x00 ACK PSH URGP=0

System log messages when this happens are inconsistent and confusing to me.  The most recent event has the following:

Wed Apr  5 05:05:26 2023 daemon.notice netifd: STARLINK (6442): udhcpc: lease of 100.89.101.233 obtained from 100.64.0.1, lease time 300
Wed Apr  5 05:06:43 2023 daemon.info uwsgi-luci: ...The work of process 17089 is done. Seeya!
Wed Apr  5 05:06:44 2023 daemon.info uwsgi-luci: Respawned uWSGI worker 2 (new pid: 23272)
Wed Apr  5 05:06:44 2023 daemon.err uwsgi[8205]: Wed Apr  5 05:06:44 2023 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Wed Apr  5 05:07:56 2023 daemon.notice netifd: STARLINK (6442): udhcpc: sending renew to server 100.64.0.1```

followed by:
Wed Apr  5 05:43:37 2023 daemon.notice netifd: STARLINK (6442): udhcpc: lease of 100.89.101.233 obtained from 100.64.0.1, lease time 300
Wed Apr  5 05:44:43 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.2.138 48:e7:da:63:bb:80
Wed Apr  5 05:44:43 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.2.138 48:e7:da:63:bb:80 HPWaterHeater
Wed Apr  5 05:44:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.2.138 48:e7:da:63:bb:80
Wed Apr  5 05:44:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.2.138 48:e7:da:63:bb:80 HPWaterHeater
Wed Apr  5 05:45:48 2023 daemon.notice netifd: Network device 'connectify0' link is down
Wed Apr  5 05:45:48 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Apr  5 05:45:48 2023 daemon.notice netifd: Interface 'wan' is now down
Wed Apr  5 05:45:48 2023 daemon.notice ttyd[10406]: rops_handle_POLLIN_netlink: DELADDR
Wed Apr  5 05:45:49 2023 daemon.debug ModemManager[30788]: hotplug: event reported: action=remove, name=connectify0, subsystem=net
Wed Apr  5 05:45:50 2023 daemon.err nginx[8752]: 2023/04/05 05:45:50 [error] 8763#0: *54301 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:45:51 2023 daemon.err nginx[8752]: 2023/04/05 05:45:51 [error] 8763#0: *54303 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:45:53 2023 daemon.err nginx[8752]: 2023/04/05 05:45:53 [error] 8763#0: *54305 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:45:54 2023 daemon.err nginx[8752]: 2023/04/05 05:45:54 [error] 8763#0: *54307 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:45:57 2023 daemon.err nginx[8752]: 2023/04/05 05:45:57 [error] 8763#0: *54311 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:46:01 2023 daemon.err nginx[8752]: 2023/04/05 05:46:01 [error] 8763#0: *54313 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.161, server: smoothwan, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:9330/", host: "192.168.2.10:9331"
Wed Apr  5 05:46:03 2023 daemon.notice netifd: Interface 'wan' is disabled

I know I have not provided enough information to resolve this issue but thought I would keep track in the hopes that we will get luck at some point and it will suddenly become helpful.

TalalMash, 2023-04-05:
Seems related to #89 
Can you generate logs to see if certain configuration is causing Speedify to crash?

TalalMash, 2023-04-05:
Note that corrupt log files crash Speedify, I have tried to mitigate that by delaying journal write to 30 seconds so data loss happens and the recovered data is not corrupted in most cases, you can remove the `mount` line in System -> Startup Local Startup, save and restart. Probably has to do with qemu virtual storage being different, is this converted VMDK or qcow2?

pbix, 2023-04-21:
I have been running HF5 for about a week.  Got another crash

```
Tue Apr 18 02:33:38 2023 daemon.notice netifd: Network device 'connectify0' link is down
Tue Apr 18 02:33:38 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Apr 18 02:33:38 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Apr 18 02:33:38 2023 daemon.err odhcp6c[12770]: Failed to send RS (Network unreachable)
Tue Apr 18 02:33:38 2023 daemon.notice ttyd[11541]: rops_handle_POLLIN_netlink: DELADDR
Tue Apr 18 02:33:38 2023 daemon.notice ttyd[11541]: rops_handle_POLLIN_netlink: DELADDR
Tue Apr 18 02:33:38 2023 daemon.err odhcp6c[12770]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Tue Apr 18 02:33:38 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Tue Apr 18 02:33:39 2023 daemon.notice netifd: Interface 'wan6' is now down
```

Not sure what "logs" you are referring to.  Most everything I have posted above is from the systemlog.  If you mean the Speedify logs I tried running "GenerateLogs.sh".   I installed bash since it is a bash script but I still get many script errors as shown below.

```
root@SmoothWAN:/usr/share/speedify# ./GenerateLogs.sh 
Generating Speedify Logs...
./GenerateLogs.sh: line 17: lsb_release: command not found
grep: /etc/issue: No such file or directory
cp: can't stat '/etc/speedify/speedify.conf': No such file or directory
./GenerateLogs.sh: line 28: journalctl: command not found
./GenerateLogs.sh: line 29: journalctl: command not found
./GenerateLogs.sh: line 30: journalctl: command not found
./GenerateLogs.sh: line 31: journalctl: command not found
BusyBox v1.35.0 (2023-03-30 12:17:21 UTC) multi-call binary.

Usage: cp [-arPLHpfinlsTu] SOURCE DEST
or: cp [-arPLHpfinlsu] SOURCE... { -t DIRECTORY | DIRECTORY }

Copy SOURCEs to DEST

        -a      Same as -dpR
        -R,-r   Recurse
        -d,-P   Preserve symlinks (default if -R)
        -L      Follow all symlinks
        -H      Follow symlinks on command line
        -p      Preserve file attributes if possible
        -f      Overwrite
        -i      Prompt before overwrite
        -n      Don't overwrite
        -l,-s   Create (sym)links
        -T      Refuse to copy if DEST is a directory
        -t DIR  Copy all SOURCEs into DIR
        -u      Copy only newer files
SpeedifyLogs/
SpeedifyLogs/dnsmasq.conf
SpeedifyLogs/HostapdService.txt
SpeedifyLogs/DnsmasqService.txt
SpeedifyLogs/SpeedifySharingService.txt
SpeedifyLogs/SpeedifyService.txt
SpeedifyLogs/summary_041823.txt
SpeedifyLogs/summary_041723.txt
SpeedifyLogs/summary_041623.txt
SpeedifyLogs/summary_041523.txt
SpeedifyLogs/summary_041423.txt
SpeedifyLogs/summary_041323.txt
SpeedifyLogs/sql.db
SpeedifyLogs/speedify_20230415_005236.log_2
SpeedifyLogs/speedify_20230415_005236.log_1
SpeedifyLogs/speedify_20230415_005236.log_0
SpeedifyLogs/speedify_20230414_222330.log_2
SpeedifyLogs/speedify_20230414_222330.log_1
SpeedifyLogs/speedify_20230414_222330.log_0
SpeedifyLogs/speedify_20230414_220636.log_0
SpeedifyLogs/speedify_20230414_214558.log_0
SpeedifyLogs/speedify_20230413_021335.log_2
SpeedifyLogs/speedify_20230413_021335.log_1
SpeedifyLogs/speedify_20230413_021335.log_0
SpeedifyLogs/settings.txt.lastGoodWrite
SpeedifyLogs/settings.txt.lastGoodRead
SpeedifyLogs/settings.txt
SpeedifyLogs/lastUsedSettings.txt
SpeedifyLogs/histogram_20230414_222330.tsv
SpeedifyLogs/histogram_20230413_021335.tsv
SpeedifyLogs/VERSION
SpeedifyLogs/13.2.0.10889_daemon_4683dce5-0981-4290-776fa597-440328ae.dmp
SpeedifyLogs/system.txt
SpeedifyLogs/OS.txt
/usr/share/speedify
./GenerateLogs.sh: line 44: who: command not found
chown: unknown group 
Log files generated at ~/SpeedifyLogs.tar.gz
root@SmoothWAN:/usr/share/speedify# 
```
Here is the generated file.
[SpeedifyLogs.tar.gz](https://github.com/SmoothWAN/SmoothWAN/files/11297577/SpeedifyLogs.tar.gz)

TalalMash, 2023-04-21:
Speedify save data is within the logs afaik, there is a generate log button in WebUI, *VPN* -> *Speedify* menu.
Not sure if the Speedify logs would be useful to read, instead, try posting the *Status* -> *System Log* -> *Kernel Log* to see if this is memory (leak) issue.
To disable the P2P logging, in terminal, type: `mv /etc/firewall.user /etc/firewall.user.disable`

pbix, 2023-04-21:
I inspected both System and Kernel logs and could see any issue related there other than what I posted above.

I do not see a "Generate Log" button on the VPN->Speedify menu.  I only see a "Download Logs & Config" button.  Will try that when it happens again.

P2P logging is disabled, thanks for that tip.

TalalMash, 2023-04-21:
```
Following that I see periodic "Objject busy errors".

Mon Mar 20 00:47:34 2023 daemon.err nlbwmon[7037]: Unable to dump conntrack: I/O error
```
This suggests storage issue with the VM, not sure where to look at.

pbix, 2023-04-22:
Up until now I had not removed any mount.  Today I commented out
"mount -o rw,remount,noatime,async,barrier=1,commit=30,errors=continue / &"

Hopefully that is the one you mean.  I will reboot shortly.

Not sure what resource could be constrained.    Below is my disk situation.
[image]

Memory allocated to the VM is 350MB and I have not seen the usage go over 290MB

TalalMash, 2023-04-22:
Added info in release page regarding RAM, 350MB is a bit too low.
Issue is similar to: 

https://smoothwan.discourse.group/t/slate-ax-cpu-to-100-and-all-network-traffic-ceased-reboot-required/57/1

https://github.com/SmoothWAN/SmoothWAN/issues/89

You can try downgrading Speedify for testing purpose in the `Options` tab. There were no reports during `12.8`.
You can also use `htop` command & press `F6` to sort by memory usage to see if other processes have abnormal memory usage.

pbix, 2023-04-28:
Based on your comments I increased the memory to 512MB.  In the last week I have not seen this issue repeat so you may be correct.  My htop looks like the below.  There is no indication that a memory limit was reached but if it was momentarily so it might not be shown.  I will keep this issue open a little while longer but I think that 512MB might be the minimum required to run this in a VM.  It seems the days when we could run OpenWRT on 16MB routers is gone.

[image]

TalalMash, 2023-04-28:
Yes, it looks like the latest Speedify version is indeed using more memory, similar issues are present on other platforms. (currently fixed using `zram` in the next release)
>  It seems the days when we could run OpenWRT on 16MB routers is gone.

Yeah, [it's increasing exponentially ](https://openwrt.org/supported_devices/432_warning#:~:text=16%2F64%20as%20an%20absolute,MB%20of%20RAM%20being%20preferred.) ;)

closed 2024-01-07 by TalalMash