SmoothWAN/SmoothWAN #144: Feature Request: Cloudflare Warp over Speedify

tiernano, 2023-04-08:
Not sure how possible this is, but would it be possible to have Cloudflare WARP (https://1.1.1.1) running on SmoothWAN and use that for sending traffic over Speedify? I know its an extra layer, but Cloudflare would have better-peering connections than Speedify, and I think I would trust Cloudflare's privacy policy more than Speedify... the WGCF tool (https://github.com/ViRb3/wgcf) allows you to generate a Wireguard config to connect to Cloudflare...

TalalMash, 2023-04-08:
Check this out: https://github.com/SmoothWAN/SmoothWAN/discussions/79#discussioncomment-4131043
You can use any VPN with PBR enabled.
You can also use split-tunneling, e.g set WG to a BYP routing table, and create a rule for LAN with a range e.g 172.17.17.0/24 for all clients. More: https://smoothwan.com/wanbyp/#split-tunneling-for-leak-proof-bypass

tiernano, 2023-04-09:
ok.... might be a stupid question, but how do I setup a wireguard VPN connection? When I click on "Status" I see "Wireguard" but its empty... and there doesn't seem to be a "Wireguard" option under "VPN"... I must be missing something simple... Thanks...

tiernano, 2023-04-09:
right... i was being stupid... i created a connection though the interface, and now have wireguard setup... I still cant figure out the PBR stuff: 
* I enabled PBR
*  under advanced config, added the wg connection to the supported interfaces
*  enabled "show chain", "show enabled" and "show protocol" columns in the web interface.
* created a policy named cloudflare, local address "172.17.17.0/24", local ports 0-65535, destination "empty" (0.0.0.0/0 fails...) remote ports 0-65535, protocol ALL, chain "PREROUTING".
* save config
* start service
* still showing as old IP from Speedify...
Did I miss something?

Thanks.

TalalMash, 2023-04-09:
Wrote a guide here, setup manually instead of PBR:
https://smoothwan.com/vpn/

tiernano, 2023-04-09:
Right... i must be doing something wrong, cause I followed those steps and still getting the old IP... Do you have screen shots of the "ipv4 rules" and "static ipv4 routes" screens? Its going to be something simple, I know, but I just cant figure out what I am doing wrong... Thnaks

TalalMash, 2023-04-09:
[image]
[image]
[image]
[image]

tiernano, 2023-04-09:
I just checked and all the config is the same... the only difference I have is I used BYP9 (swapped it to BYP1 and no difference) and MTU on WARP was set to 1280, per the WGCF config (removed that, but still no difference). No other config differences... running on a Flint, the latest install, and IPs on the internet are Static... weird... I will try a reinstall of SmoothWAN tomorrow and see if there is a weird config issue somewhere else I am missing... ill get back to you then.

TalalMash, 2023-04-18:
Thanks for testing, I did notice that as well, it happens to choose WANs on first up, and setting `netifd` to bind to `connectify0` did not work for some reason, WireGuard doesn't follow the default routing table on OpenWrt.
I've added extra instructions in the meantime using firewall markings method:
https://smoothwan.com/vpn/ - *WireGuard extra steps for v0.99.9*

I've also noticed that WARP is throttled behind data center IP address like Speedify, need to do more testing to confirm.

TalalMash, 2023-04-18:
You can also try in terminal: `ip rule add iif lo sport 2408 lookup main` (replace 2408 with your WG peer port)
Then restart the WG interface in LuCI.

closed 2023-12-25 by TalalMash