SmoothWAN/SmoothWAN #147: Support for NanoPi R6S

TalalMash, 2023-05-04:
**Status:** In development
Current source: official FriendlyWrt image builder build.

TalalMash, 2023-05-05:
@kputland https://github.com/SmoothWAN/SmoothWAN/issues/95#issuecomment-1533787653, uploaded a test release: https://github.com/SmoothWAN/SmoothWAN/releases/tag/local1

kputland, 2023-05-05:
sd-card image?

kputland, 2023-05-05:
It boots from image installed on sd-card.

It doesn't want to flash to eMMC
[image]

default password remained password though adguard was the smoothwan default
USB network didn't get dhcp on reconnect.

speedify install logs show  installed, but nothing is in /usr/share/speedify
[image]
```
/
root@FriendlyWrt:/# cd /usr/share/speedify/
root@FriendlyWrt:/usr/share/speedify# ls
logs
root@FriendlyWrt:/usr/share/speedify# ls -l
total 4
drwxr-xr-x 2 root root 4096 May  5 12:40 logs
root@FriendlyWrt:/usr/share/speedify#

```

speedify UI says not configured.

for cpufreq and irqbalance, the /etc/config files should be used
```
# from rc.local
#Set CPU to performance (some CPUs stutter when changing freq)
echo "performance" > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor &

#Run irqbalance
irqbalance &
```
OR
```
for x in  /sys/devices/system/cpu/cpufreq/policy*/scaling_governor; do echo "performance" > $x; done
```

uci settings are there for speedifyconf
```
root@FriendlyWrt:~# uci show speedifyconf
speedifyconf.Setup=config
speedifyconf.Setup.enabled='1'
speedifyconf.Setup.apt='https://apt.connectify.me/'
speedifyconf.Setup.logpath='logs'
speedifyconf.Setup.renamer='1'
speedifyconf.Setup.version='13.2.1-10893'
root@FriendlyWrt:~#
```

```
root@FriendlyWrt:~# find / -name '*speedify*'
/etc/rc.d/S99speedifyconf
/etc/init.d/speedifyconf
/etc/config/speedifyconf
/www/luci-static/resources/view/speedifyconf
/usr/share/rpcd/acl.d/luci-app-speedifyconf.json
/usr/share/luci/menu.d/luci-app-speedifyconf.json
/usr/share/speedify
/usr/lib/lua/luci/view/speedifyconf
/usr/lib/lua/luci/view/admin_status/index.htm.speedify
/usr/lib/lua/luci/model/cbi/speedifyconf
/usr/lib/lua/luci/controller/speedifyconf
/usr/lib/opkg/info/luci-app-speedifyconf.control
/usr/lib/opkg/info/luci-app-speedifyconf.list
/usr/lib/opkg/info/luci-app-speedifyconf.prerm
/usr/lib/opkg/info/speedifyconf.conffiles
/usr/lib/opkg/info/speedifyconf.control
/usr/lib/opkg/info/speedifyconf.list
/usr/lib/opkg/info/speedifyconf.prerm
/usr/lib/speedifyconf
/usr/lib/speedifyconf/speedifyconfig.log
/tmp/speedifyconfig.log
/tmp/.uci/speedifyconf
/tmp/lock/procd_speedifyconf.lock
/tmp/run/rpcd/uci-2b87409dbe88683d21283d1dbdef1a67/speedifyconf
root@FriendlyWrt:~#
```

Let me know what you need

kputland, 2023-05-05:
```

sh: can't open 'DisableRpFilter.sh': No such file or directory
uci: Entry not found
sh: 1: unknown operand
nice: ‘capsh’: No such file or directory
/etc/rc.common: ./run.sh: line 96: ./speedify_cli: not found
Running
Syntax: ./run.sh [command]

Available commands:
        start           Start the service
        stop            Stop the service
        restart         Restart the service
        reload          Reload configuration files (or restart if service does not implement reload)
        enable          Enable service autostart
        disable         Disable service autostart
        enabled         Check if service is started on boot

root@FriendlyWrt:/usr/lib/speedifyconf# find / -name ^C
root@FriendlyWrt:/usr/lib/speedifyconf# find / -name DisableRpFilter.sh
root@FriendlyWrt:/usr/lib/speedifyconf#
```

kputland, 2023-05-05:
capsh required `libcap-bin` and `libcap-ng-bin`

kputland, 2023-05-05:
ar missing so deb never extracted.

kputland, 2023-05-05:
Manually got it all installed.
eth2 2.5Gb WAN port is not configured by default

[image]

[image]

kputland, 2023-05-05:
Moved the hotspot onto a usb3 hub and the interface name changed
[image]

kputland, 2023-05-05:
eth2 (Comcast) has connectivity issues if speedify changes connections to a different server.

kputland, 2023-05-05:
I can definitely work with this kind of speeds.
[image]

kputland, 2023-05-05:
900Mbps

[image]

kputland, 2023-05-05:
```
root@FriendlyWrt:/sys/class/leds# /etc/init.d/led start
setting up led WAN
setting up led LAN1
setting up led LAN2
root@FriendlyWrt:/sys/class/leds# /etc/init.d/led enabled
root@FriendlyWrt:/sys/class/leds#
```

kputland, 2023-05-05:
weird interface state
[image]
```
root@FriendlyWrt:/sys/class/leds# ping -I USB_c146 192.168.2.1
PING 192.168.2.1 (192.168.2.1): 56 data bytes
64 bytes from 192.168.2.1: seq=0 ttl=64 time=2.292 ms
64 bytes from 192.168.2.1: seq=1 ttl=64 time=2.069 ms
64 bytes from 192.168.2.1: seq=2 ttl=64 time=2.106 ms
^C
--- 192.168.2.1 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 2.069/2.155/2.292 ms
root@FriendlyWrt:/sys/class/leds#
```

TalalMash, 2023-05-05:
Thanks! Looks promising, gigabit is impressive, and this is on public servers.
Will release the EMMC image next time and Docker enabled too.

- Info before next image:
```
uci show
service
ip route
ls /sys/devices/system/cpu
cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq
```
- For the `ip route` try to replicate the behaviour when eth2 is not working.
- Regarding `cpufreq`, do you mean the FriendlyWrt CPU frequency LuCI app should replace it?
`- ...cpuinfo_cur_freq` should show if the `for` loop is needed for small/large core adjustment.
> Moved the hotspot onto a usb3 hub and the interface name changed

This is the expected behaviour.
> USB network didn't get dhcp on reconnect.

Does it work with the renamer off and set manually in *Network* -> *Interfaces*?

- What's under the *NAS* menu?

kputland, 2023-05-05:
under NAS. 
[image]
Though in the default FriendlyWRT there were six or seven choices.

kputland, 2023-05-05:
[smoothwan_r6s_logs.zip](https://github.com/SmoothWAN/SmoothWAN/files/11409081/smoothwan_r6s_logs.zip)

irqbalance can be run from the service enabled, not from rc.local.

cpufreq I think will need the loop. luci/uci doesn't know about the different processor pools. The luci/uci settings only affected policy0 and left the performance cores at a lower frequence, unless it's smart enough to sort it all out. I'd have to do back-to-back tests with the governor changes.

[image]

kputland, 2023-05-05:
interfaces are still renamed with usb renaming checkbox unchcecked.

kputland, 2023-05-05:
thermal performance vs schedutil
[image]

kputland, 2023-05-05:
clicking restart interface in the network menu doesn't bring the interface back up.
log attached
[usb_network_interface_restart.log](https://github.com/SmoothWAN/SmoothWAN/files/11409737/usb_network_interface_restart.log)

TalalMash, 2023-05-06:
Added branch and new release: 
https://github.com/SmoothWAN/SmoothWAN/releases/tag/local1
https://github.com/SmoothWAN/SmoothWAN/tree/npi-r6s/devconfigs/npir6s

TalalMash, 2023-05-06:
> clicking restart interface in the network menu doesn't bring the interface back up. log attached [usb_network_interface_restart.log](https://github.com/SmoothWAN/SmoothWAN/files/11409737/usb_network_interface_restart.log)

You may need to restart the device, this also happens with other hardware.

TalalMash, 2023-05-06:
Uploaded new image with pre-made `usb0` files and `uhttpd` fix.

kputland, 2023-05-06:
Not sure what's going on here. eth2 is assigned to wan and wan6 as well as AutoWANeth2
[image]

kputland, 2023-05-06:
Causing two udhcpc instances 
```
12202 root      1320 S    udhcpc -p /var/run/udhcpc-eth2.pid -s /lib/netifd/dhcp.script -f -t 0 -i eth2 -x hostname:FriendlyWrt -C -R -O 121
12205 root      1320 S    udhcpc -p /var/run/udhcpc-eth2.pid -s /lib/netifd/dhcp.script -f -t 0 -i eth2 -x hostname:FriendlyWrt -C -R -O 121

```

kputland, 2023-05-06:
speedifyui fails to download and install.
Manual installation and it's now running.
```
root@FriendlyWrt:/tmp/spddw/speedifyui# UIDW=$(sed -n '/speedifyui/{nnnnnnnn;p;q}' $PKGS | awk '/Filename/{gsub("Filename: ", "");print;exit}')
sed: -e expression #1, char 15: extra characters after command
root@FriendlyWrt:/tmp/spddw/speedifyui# 
```

had to change it to this 
```
root@FriendlyWrt:/tmp/spddw/speedifyui# sed -n '/speedifyui/{n;n;n;n;n;n;n;n;p;q}' $PKGS
Filename: pool/main/s/speedifyui/speedifyui_13.2.1-10893_arm64.deb
```
Or this could work
```
grep 'Filename.*speedifyui' $PKGS | head -n1 | sed -n "s/Filename: //p"
```
```
root@FriendlyWrt:/tmp/spddw/speedifyui# sed --version
sed (GNU sed) 4.8
```

This nnnnn is in two other places in `run.sh`

kputland, 2023-05-06:
`uci-defaults/99_custom` didn't appear to make it into the image. `smoothwan_0.99.5HF5_4296f32_friendlywrt_22.03_20230506_nanopi-r6s_arm64_sd`

```
root@FriendlyWrt:~# ls -l /etc/uci-defaults/
total 4
-rwxrwxr-x 1 root root 2938 Mar 10 16:34 99_luci-app-emmc-tools
root@FriendlyWrt:~# 
```

kputland, 2023-05-06:
eth0, though assisgned to AutoWANeth1 is also still part of the br-lan. In my case I need to leave it as attached to br-lan.
Maybe eth0 should be left unassigned with documentation to add it back for LAN or WAN use.
```
-rwxrwxr-x 1 root root 2938 Mar 10 16:34 99_luci-app-emmc-tools
root@FriendlyWrt:~# uci show network.@device[1]
network.cfg040f15=device
network.cfg040f15.name='br-lan'
network.cfg040f15.type='bridge'
network.cfg040f15.ports='eth1' 'eth0'
```

kputland, 2023-05-06:
Something with the eth2 config looks to have some bridge holdover. It souldn't have the `list ports eth1`
```
network.cfg030f15=device
network.cfg030f15.name='eth2'
network.cfg030f15.macaddr='52:53:d4:77:f2:cc'
network.cfg030f15.ports='eth1'
```

TalalMash, 2023-05-06:
Thanks, seems like `sed` is full in friendlywrt.
Upon error free exit, OpenWrt deletes the `uci-defaults` contents. 
> Not sure what's going on here. eth2 is assigned to wan and wan6 as well as AutoWANeth2

Forgot to remove these from bridge, also it seems like the USB profiles are missing, guess I uploaded the old image.

TalalMash, 2023-05-06:
Uploaded new one with rename to PortOne/PortTwo.

kputland, 2023-05-06:
None of the network config happened.
[image]

kputland, 2023-05-06:
```

root@FriendlyWrt:/etc# ls -l /etc/uci-defaults/
total 12
-rw-rw-r-- 1 root root 5717 Apr  9 20:27 99-custom
-rwxrwxr-x 1 root root 2938 Mar 10 16:34 99_luci-app-emmc-tools
root@FriendlyWrt:/etc# 

```

kputland, 2023-05-06:
```root@FriendlyWrt:/etc/uci-defaults# bash 99-custom 
99-custom: command substitution: line 2: unexpected EOF while looking for matching `''
99-custom: command substitution: line 2: unexpected EOF while looking for matching `''
cfg11dc81
cfg12ad58
cfg13ad58
cfg14ad58
99-custom: line 140: /etc/init.d/usbmuxd: No such file or directory
99-custom: line 143: /etc/init.d/usbmuxd: No such file or directory
root@FriendlyWrt:/etc/uci-defaults# /usr/bin/chattr: Not supported while reading flags on /etc/resolv.conf
passwd: password for root changed by root

```

TalalMash, 2023-05-06:
woops typo, replace backtick with `'` in 99-custom firewall section near AutoUSB0

kputland, 2023-05-06:
several lines with the type. Fixed and rebooted. 99-custom is now gone from uci-defaults. This remains
```
root@SmoothWAN:~# ls /etc/uci-defaults/
99_luci-app-emmc-tools
root@SmoothWAN:~# 
```

kputland, 2023-05-06:
If eth0 is PortOne and part of wan, then need to set it's MAC as it comes by default with same MAC applied to br-lan, eth0, and eth1

TalalMash, 2023-05-06:
Interesting, OpenWrt should automatically do that, can you post `uci show`?

kputland, 2023-05-06:
uhttpd is still running instead of nginx
```
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2231/uhttpd

```

kputland, 2023-05-06:
[uci.log](https://github.com/SmoothWAN/SmoothWAN/files/11413394/uci.log)

TalalMash, 2023-05-06:
`\etc\board.d\02_network` has few pointers regarding MAC address assignment.
Try changing `ucidef_set_interface_lan 'eth0'` to `eth1` under r6s setup_interfaces.

TalalMash, 2023-05-06:
> uhttpd is still running instead of nginx
> 
> ```
> tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2231/uhttpd
> ```

Weird, enabling `luci-ssl-nginx` next release, you could also try installing that, but it may break configuration, not sure.

kputland, 2023-05-06:
nginx is intalled. I just did
```
/etc/init.d/uhttpd stop
/etc/init.d/nginx start
/etc/init.d/uhttpd disable
```
UI is working now and speedify UI proxy is working now.

kputland, 2023-05-06:
[image]

kputland, 2023-05-06:
checkbox for USB naming doesn't appear to change anything. On or off and devices are still renamed.

TalalMash, 2023-05-06:
> checkbox for USB naming doesn't appear to change anything. On or off and devices are still renamed.

Even after a reboot?

kputland, 2023-05-06:
Didn't try a reboot... but missed the step about restart speedify. Forgot that's processed in the speedifyconf init. However need the AutoUSBeth{3..5} configured too.

TalalMash, 2023-05-06:
Ah I see, some of the USB devices are `eth*`? Will add these like the x86 version next release then.

kputland, 2023-05-06:
The AutoUSBethXX are already in the SlateAX, or I configured them that way. I can't recall.

kputland, 2023-05-06:
There's a hiccup in the firewall Zones
[image]

kputland, 2023-05-06:
Hmmm... Doesn't appear to be forwarding traffic 
speedify disabled, ping from the npir6s
```
root@SmoothWAN:/tmp/spddw/speedifyui# 
root@SmoothWAN:/tmp/spddw/speedifyui# ip route
default via 192.168.1.1 dev eth3 proto static src 192.168.1.7 metric 33 
default via 192.168.2.1 dev eth4 proto static src 192.168.2.6 metric 34 
172.17.17.0/24 dev br-lan proto kernel scope link src 172.17.17.2 
192.168.1.0/24 dev eth3 proto static scope link metric 33 
192.168.2.0/24 dev eth4 proto static scope link metric 34 
root@SmoothWAN:/tmp/spddw/speedifyui# ping 4.2.2.1`
> ^C
root@SmoothWAN:/tmp/spddw/speedifyui# ping 4.2.2.1
PING 4.2.2.1 (4.2.2.1): 56 data bytes
64 bytes from 4.2.2.1: seq=0 ttl=50 time=52.352 ms
64 bytes from 4.2.2.1: seq=1 ttl=50 time=51.153 ms
64 bytes from 4.2.2.1: seq=2 ttl=50 time=53.783 ms
^C
--- 4.2.2.1 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 51.153/52.429/53.783 ms
root@SmoothWAN:/tmp/spddw/speedifyui# 

```

Now from my computer
```
kputland@VONWQG1QWGJ21 Downloads  % ping 4.2.2.1 
PING 4.2.2.1 (4.2.2.1): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
^C
--- 4.2.2.1 ping statistics ---
5 packets transmitted, 0 packets received, 100.0% packet loss
```

```
kputland@VONWQG1QWGJ21 Downloads  % netstat -rn
Routing tables

Internet:
Destination        Gateway            Flags           Netif Expire
default            172.17.17.2        UGScg             en9       
127                127.0.0.1          UCS               lo0       
127.0.0.1          127.0.0.1          UH                lo0       
169.254            link#24            UCS               en9      !
172.17.17/24       link#24            UCS               en9      !
172.17.17.2/32     link#24            UCS               en9      !
172.17.17.2        52:53:d4:77:f2:cd  UHLWIir           en9   1175
172.17.17.108/32   link#24            UCS               en9      !
224.0.0/4          link#24            UmCS              en9      !
224.0.0.251        1:0:5e:0:0:fb      UHmLWI            en9       
239.255.255.250    1:0:5e:7f:ff:fa    UHmLWI            en9       
255.255.255.255/32 link#24            UCS               en9      !
```

kputland, 2023-05-06:
weird. WAN=>ACCEPT (Forward) was set to reject.

TalalMash, 2023-05-06:
> There's a hiccup in the firewall Zones [image]

This is because the `99-custom` script ran twice after replacing the backtick.

Will set to ACCEPT next release, seems like it was REJECT due to `fullcone` enabled by FriendlyWrt.

kputland, 2023-05-06:
Seems like there's still a timing problem with the 2.5Gbps ethernet ports and any USB connected ports. If the USB device is connected during power-on it was assigned eth1 instead of the onboard ports. I might have to wait to test this. I've configured the device as the network gateway for now. I'll probably order a second.

kputland, 2023-05-06:
[image]

TalalMash, 2023-05-06:
Seems like we have to delay the USB module.

kputland, 2023-05-07:
OOM error from nlbwmon

```

Sun May  7 07:07:20 2023 daemon.info vnstatd[12675]: Info: Traffic rate for "eth2" higher than set maximum 1000 Mbit (20s->2673868800, r5082038657 t279285, 64bit:1), syncing.
Sun May  7 07:07:40 2023 daemon.info vnstatd[12675]: Info: Traffic rate for "eth2" higher than set maximum 1000 Mbit (20s->2673868800, r13552066019 t331342, 64bit:1), syncing.
Sun May  7 07:08:45 2023 daemon.err nlbwmon[1390]: Netlink receive failure: **Out of memory**
Sun May  7 07:08:45 2023 daemon.err nlbwmon[1390]: Unable to dump conntrack: No buffer space available

```

```
root@SmoothWAN:~# free
              total        used        free      shared  buff/cache   available
Mem:        7852280      301424     7179484       39284      371372     7415512
Swap:             0           0           0

```

TalalMash, 2023-05-07:
https://github.com/jow-/nlbwmon/issues/43

> OOM error from nlbwmon
>
> ```
>
> Sun May  7 07:07:20 2023 daemon.info vnstatd[12675]: Info: Traffic rate for "eth2" higher than set maximum 1000 Mbit (20s->2673868800, r5082038657 t279285, 64bit:1), [...]

TalalMash, 2023-05-07:
Try moving the USB module to prevent it from loading, then load it later, seems like Rockchip ethernet assignment is delayed for some reason.
```
mv /etc/modules.d/usb-net /root
sed -i '1imodprobe usbnet' /etc/rc.local
```

kputland, 2023-05-08:
I have another R6S coming to make testing easier. Though I'm interested in the BPI-R3. Trying to decide if I keep the Slate AX at this point. I have about another week to return it.

kputland, 2023-05-08:
collectd also overeporting 1.7P in last 24hr
[image]

Nothing odd in bridging 
```
root@SmoothWAN:/usr/share/speedify# brctl show
bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.5253d477f2cd	no		eth0
							eth5
							eth1
root@SmoothWAN:/usr/share/speedify# 
```
Interfaces
[image]

[36 more replies]