SmoothWAN/SmoothWAN #150: Upgrade (or remove) AdGuardHome?

jhenkens, 2023-06-09:
Hi! AdGuardHome has a bug in older versions (https://github.com/AdguardTeam/AdGuardHome/issues/5281) that sets the MTU _really_ low. This has very subtle bugs, including breaking Tailscale subnet routing. It looks like once could override the default AdGuardHome config at `/etc/adguardhome.yaml`, or we can upgrade the AdGuardHome package, but I can't see that available in the OpenWRT repositories.

TalalMash, 2023-06-09:
Do you mean ADG is changing the br-lan MTU? I couldn't replicate the issue on the latest firmware, if you're running an older SmoothWAN version then this may be a problem since ADG used to be the DHCP server.

jhenkens, 2023-06-09:
> Do you mean ADG is changing the br-lan MTU? I couldn't replicate the issue on the latest firmware, if you're running an older SmoothWAN version then this may be a problem since ADG used to be the DHCP server.

Ah, didn't realized it changed back to not being the DHCP server. My mistake. I haven't updated in a bit as I've got to set some time aside to copy settings to/from.

TalalMash, 2023-06-09:
Great, updating next version to static binary release instead of OpenWrt repo version due to other missing features as well.