SmoothWAN/SmoothWAN #151: Local network through usb ethernet dongle

zbynekdrlik, 2023-06-16:
Hi I have laptop without any hw ethernet port. Have to use usb ethernet dongle for local ethernet also? 
Normally it seams that all usb ethernets are used for wans.

TalalMash, 2023-06-16:
Hi, yes you need to disable USB network interface renamer in Speedify menu (under VPN).
Restart, then it should be automatically assigned to LAN, else add it in: Network -> Interfaces -> Devices -> Edit next to `br-lan`

zbynekdrlik, 2023-06-17:
I am not able to open web configuration as there is no ethernet connection due missing eth0 device. Could I do this with shell?
I am mainly interested in second option as i will need multiple usbeth dongles.

TalalMash, 2023-06-17:
Try changing `renamer` to 0 in: `/etc/config/speedifyconf` then restart.
`ip a` to view network adapters
`/etc/config/network` shows br-lan default port which is `eth0`.
`dmesg | grep eth` should generally show what interface is connected to.

closed 2023-12-25 by TalalMash