SmoothWAN/SmoothWAN #155: AdGuard Disable

pbix, 2023-06-25:
How can I prevent any AdGuard software from running on my SmoothWAN router.  I do not use this feature and never configured it.  But it still spams the logs with messages.  

I see adguardhome in the installed software list.  Is it safe to remove this and/or any other modules to prevent this software from running?

Example log spam below:

```
Sun Jun 25 09:20:41 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339367 [info] Received signal "terminated"
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339403 [info] stopping AdGuard Home
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339417 [info] stopping http server...
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339670 [info] stopped http server
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339766 [info] Stopping the DNS proxy server
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.339886 [info] Stopped the DNS proxy server
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[19436]: 2023/06/25 13:20:41.371836 [info] stopped
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.424818 [info] AdGuard Home, version v0.107.21
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.425113 [info] AdGuard Home updates are disabled
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.429855 [info] tls: using default ciphers
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.460403 [info] Initializing auth module: /etc/adguardhome/data/sessions.db
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.460605 [info] auth: initialized.  users:1  sessions:0
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.460634 [info] web: initializing
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.631338 [info] dnsproxy: cache: enabled, size 4096 b
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.631696 [info] MaxGoroutines is set to 300
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.636865 [info] AdGuard Home is available at the following addresses:
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640580 [info] Go to http://127.0.0.1:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640654 [info] Go to http://[::1]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640670 [info] Go to http://192.168.2.1:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640686 [info] Go to http://[fe80::ccc7:7fff:fe98:11ad%br-lan]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640698 [info] Go to http://198.18.61.158:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640711 [info] Go to http://[fe80::ccc7:7fff:fe98:11ad%eth0.3]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640724 [info] Go to http://100.90.224.85:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640740 [info] Go to http://[fd7a:115c:a1e0:ab12:4843:cd96:625a:e055]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640754 [info] Go to http://[fe80::5a45:f8e8:3ba9:3354%br-tailscale0]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640767 [info] Go to http://10.202.0.2:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640781 [info] Go to http://[fe80::e5b3:3c8b:c62c:697f%connectify0]:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640792 [info] Go to http://100.89.101.233:3000
Sun Jun 25 09:20:41 2023 daemon.err AdGuardHome[20132]: 2023/06/25 13:20:41.640804 [info] Go to http://[fe80::ccc7:7fff:fe98:11ad%eth0.2]:3000
```

TalalMash, 2023-06-25:
ADG is only restarted on network changes or Speedify restart, the spam suggests a system issue.
It's currently used as the main DNS resolver in a disabled state for one click configuration.
You can remove it in package management page, and set the DNS option in DHCP setting to a public server like 8.8.8.8 in *Interfaces* -> *LAN*-> Edit -> DHCP server -> Adv. Settings -> DHCP-options -> "6,8.8.8.8".
Power cycle or reconnect all connected clients.

pbix, 2023-06-25:
Which packages to remove?

adhomeconfig?
adguardhome?

pbix, 2023-06-30:
I removed only adguardhome and it seems to do the trick.  No more spam in the logs.

closed 2023-06-30 by pbix