SmoothWAN/SmoothWAN #156: Second interface does not come up automatically on reboot

pbix, 2023-06-30:
On my router I have two interfaces for SmoothWAN & Speedify to use.  The two interfaces are eth0.2 & eth0.3.

On reboot I observe that eth0.2 is not being used by Speedify, only eth0.3 is being used.  If I stop and then restart eth0.2 Speedify starts using it.  I need some ideas how to debug this issue. 

Below is the syslog starting from a reboot.  Near the end of the log I manually stop and restart eth0.2 and Speedify begins to work properly.  The log is quite long I know, all but about the last 20 lines are part of the initial boot.  About 20 lines from the end you can see where eth0.2 is stopped and restarted.

Some text removed at the top due to character limits in this forum.

Any help would be appreciated.

```
Fri Jun 30 17:14:14 2023 kern.notice kernel: [    0.000000] Linux version 5.10.161 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20028-43d71ad93e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Jan 3 00:24:21 2023
Fri Jun 30 17:14:14 2023 kern.info kernel: [   10.940737] e100: Intel(R) PRO/100 Network Driver
Fri Jun 30 17:14:14 2023 kern.info kernel: [   10.945156] e100: Copyright(c) 1999-2006 Intel Corporation
Fri Jun 30 17:14:14 2023 kern.info kernel: [   10.955743] e1000e: Intel(R) PRO/1000 Network Driver
Fri Jun 30 17:14:14 2023 kern.info kernel: [   10.960318] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
Fri Jun 30 17:14:15 2023 daemon.err ModemManager[4722]: hotplug: parent device sysfspath not found
Fri Jun 30 17:14:15 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jun 30 17:14:15 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jun 30 17:14:15 2023 daemon.info dnsmasq[1]: started, version 2.86 DNS disabled
Fri Jun 30 17:14:15 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Fri Jun 30 17:14:15 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Jun 30 17:14:15 2023 authpriv.info dropbear[5193]: Not backgrounding
Fri Jun 30 17:14:16 2023 user.notice pbr: Reloading pbr due to includes of firewall
Fri Jun 30 17:14:16 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Fri Jun 30 17:14:16 2023 daemon.notice procd: /etc/rc.d/S19firewall: Command failed: ubus call service delete { "name": "miniupnpd" } (Not found)
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'lan' is enabled
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'lan' is setting up now
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'lan' is now up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: bridge 'br-lan' link is up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'InfiniSky' is enabled
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'Starlink' is enabled
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'Starlink' is setting up now
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'Starlink' is now up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'loopback' is enabled
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'loopback' is now up
Fri Jun 30 17:14:18 2023 daemon.debug ModemManager[6418]: hotplug: event reported: action=add, name=SQM_IFB_c930d, subsystem=net
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: VLAN 'eth0.3' link is up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'InfiniSky' has link connectivity
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'InfiniSky' is setting up now
Fri Jun 30 17:14:18 2023 daemon.notice netifd: VLAN 'eth0.2' link is up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'Starlink' has link connectivity
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Network device 'lo' link is up
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Fri Jun 30 17:14:18 2023 daemon.err ModemManager[6418]: hotplug: parent device sysfspath not found
Fri Jun 30 17:14:18 2023 daemon.notice netifd: InfiniSky (6484): udhcpc: started, v1.35.0
Fri Jun 30 17:14:18 2023 daemon.notice netifd: InfiniSky (6484): udhcpc: broadcasting discover
Fri Jun 30 17:14:18 2023 daemon.debug ModemManager[6485]: hotplug: event reported: action=remove, name=SQM_IFB_c930d, subsystem=net
Fri Jun 30 17:14:18 2023 daemon.notice netifd: InfiniSky (6484): udhcpc: broadcasting select for 198.18.61.189, server 198.18.61.1
Fri Jun 30 17:14:18 2023 daemon.debug ModemManager[6557]: hotplug: event reported: action=add, name=br-lan, subsystem=net
Fri Jun 30 17:14:18 2023 daemon.info vnstatd[6621]: Info: vnStat daemon 2.9 started. (pid:6621 uid:0 gid:0 64-bit)
Fri Jun 30 17:14:18 2023 daemon.err ModemManager[6557]: hotplug: parent device sysfspath not found
Fri Jun 30 17:14:18 2023 daemon.notice netifd: InfiniSky (6484): udhcpc: lease of 198.18.61.189 obtained from 198.18.61.1, lease time 28800
Fri Jun 30 17:14:18 2023 daemon.notice netifd: Interface 'InfiniSky' is now up
Fri Jun 30 17:14:18 2023 daemon.debug ModemManager[6645]: hotplug: event reported: action=add, name=eth0.3, subsystem=net
Fri Jun 30 17:14:18 2023 daemon.err ModemManager[6645]: hotplug: parent device sysfspath not found
Fri Jun 30 17:14:19 2023 daemon.info [6758]: <info>  ModemManager (version 1.18.6) starting in system bus...
Fri Jun 30 17:14:19 2023 daemon.debug ModemManager[6780]: hotplug: event reported: action=add, name=eth0.2, subsystem=net
Fri Jun 30 17:14:19 2023 daemon.err ModemManager[6780]: hotplug: parent device sysfspath not found
Fri Jun 30 17:14:19 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: [uWSGI] getting INI configuration from /etc/uwsgi/emperor.ini
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: *** Starting uWSGI 2.0.20 (64bit) on [Fri Jun 30 17:14:19 2023] ***
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: compiled with version: 11.2.0 on 30 March 2023 12:17:21
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: os: Linux-5.10.161 #0 SMP Tue Jan 3 00:24:21 2023
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: nodename: SmoothWAN
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: machine: x86_64
Fri Jun 30 17:14:19 2023 daemon.err uwsgi[6950]: clock source: unix
[... 420 lines cut ...]
Fri Jun 30 17:36:41 2023 daemon.err uwsgi[6950]: *** has_emperor mode detected (fd: 8) ***
Fri Jun 30 17:36:41 2023 daemon.err uwsgi[6950]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-cgi_io.ini
Fri Jun 30 17:36:41 2023 daemon.err uwsgi[6950]: Fri Jun 30 17:36:41 2023 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini has been spawned
Fri Jun 30 17:36:41 2023 daemon.err uwsgi[6950]: Fri Jun 30 17:36:41 2023 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is ready to accept requests
Fri Jun 30 17:36:41 2023 daemon.err uwsgi[6950]: Fri Jun 30 17:36:41 2023 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is now loyal
Fri Jun 30 17:37:01 2023 daemon.notice netifd: Interface 'Starlink' is now down
Fri Jun 30 17:37:01 2023 daemon.notice ttyd[10687]: rops_handle_POLLIN_netlink: DELADDR
Fri Jun 30 17:37:01 2023 daemon.notice ttyd[10687]: rops_handle_POLLIN_netlink: DELADDR
Fri Jun 30 17:37:01 2023 daemon.debug ModemManager[18097]: hotplug: event reported: action=remove, name=eth0.2, subsystem=net
Fri Jun 30 17:37:01 2023 daemon.notice netifd: Interface 'Starlink' is disabled
Fri Jun 30 17:37:01 2023 daemon.notice netifd: VLAN 'eth0.2' link is down
Fri Jun 30 17:37:01 2023 daemon.notice netifd: Interface 'Starlink' has link connectivity loss
Fri Jun 30 17:37:11 2023 daemon.notice netifd: Interface 'Starlink' is enabled
Fri Jun 30 17:37:11 2023 daemon.notice netifd: Interface 'Starlink' is setting up now
Fri Jun 30 17:37:11 2023 daemon.notice netifd: Interface 'Starlink' is now up
Fri Jun 30 17:37:11 2023 daemon.notice netifd: VLAN 'eth0.2' link is up
Fri Jun 30 17:37:11 2023 daemon.notice netifd: Interface 'Starlink' has link connectivity
Fri Jun 30 17:37:11 2023 daemon.debug ModemManager[18431]: hotplug: event reported: action=add, name=eth0.2, subsystem=net
[... cut ...]
Fri Jun 30 17:37:12 2023 user.notice firewall: Reloading firewall due to ifup of Starlink (eth0.2)
Fri Jun 30 17:37:12 2023 user.notice pbr: Reloading pbr due to includes of firewall
Fri Jun 30 17:37:12 2023 user.notice pbr: Reload on firewall action aborted: service not running.
Fri Jun 30 17:37:13 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of Starlink (eth0.2)
Fri Jun 30 17:37:14 2023 daemon.info [6758]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:12.0': not supported by any plugin
``

TalalMash, 2023-06-30:
Hi! Post your `/etc/config/network` to check more info with private info redacted if any.

pbix, 2023-07-01:
Please the attached and thanks for taking a look at this issue for me.

```

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.2.1'

config interface 'Starlink'
	option device 'eth0.2'
	option proto 'static'
	option netmask '255.192.0.0'
	option gateway '100.xx.0.1'

config interface 'InfiniSky'
	option proto 'dhcp'
	option device 'eth0.3'

config route
	option interface 'Starlink'
	option target '192.168.100.1/24'

config interface 'wan'
	option force_link '0'
	option proto 'static'
	option device 'connectify0'
	option ipaddr '10.202.0.xx'
	option netmask '255.255.255.0'
	option gateway '10.202.0.1'

config interface 'wan6'
	option proto 'dhcpv6'
	option device 'connectify0'
	option reqaddress 'try'
	option reqprefix 'auto'

```

TalalMash, 2023-09-28:
Hi I missed your post notification, try doing vlan over br-lan, e.g br-lan.3 , br-lan.2 etc..

TalalMash, 2023-12-25:
Possible misconfiguration, re-opening if issue persists.

closed 2023-12-25 by TalalMash