SmoothWAN/SmoothWAN #167: SmoothWAN 1.0 release progress

TalalMash, 2023-09-22:
[Currently held by:](https://github.com/orgs/SmoothWAN/projects/2/views/1)
- GL.iNet official builder repository [has gone private](https://forum.gl-inet.com/t/formal-statement-on-gl-inet-software-going-closed-source/33288/16), waiting for updated imagebuilder repo (wlan-ap/qualcomm-sdk)
- Speedify is releasing official OpenWrt supported package 🎊

bencorrado, 2023-09-22:
> Speedify is releasing official OpenWrt supported package 🎊

Where did you see this announcement?  It is great news!

TalalMash, 2023-09-22:
> > Speedify is releasing official OpenWrt supported package 🎊
> 
> Where did you see this announcement? It is great news!

The past three live streams, there is no exact ETA but it is mentioned by the end of the year.

spacesuitdiver, 2023-12-15:
@TalalMash it's getting close to the end of the year eeeek hehe... I'm curious how is the current implementation different?

stuxamber, 2024-01-15:
End of year have passed. Are we there yet?

TalalMash, 2024-01-15:
Hi! @stuxamber Any long awaited feature?

TalalMash, 2024-01-15:
> @TalalMash it's getting close to the end of the year eeeek hehe... I'm curious how is the current implementation different?

From the past livestreams, it seems to be performance related, I hope UDP transport mode is improved since it seems to be very slow on the Slate/Flint.

jmdurant, 2024-03-20:
from today's livestream, it sounds like late summer for openwrt speedify launch (after they launch with partner?)