SmoothWAN/SmoothWAN #168: Add kmod-fs-exfat

leaberry, 2023-09-27:
I tried following the directions to compile my own kernel, but it's just not working (go errors on adguard). The default GL.inet kernel had exfat support, would be great to have that on Smoothwan too. 

And a big thanks for the project, it's very neat and much appreciated for those of us with multiple of internet options, but none too reliable. I use starlink/LTE combined to make zoom calls excellent for my day job.

TalalMash, 2023-09-28:
Added first image builder configuration as test, this use precompiled glinet packages and kernel, which allows you to install from glinet repositories.
The original glinet-infra-builder repo has gone private.
Compiled [test image](https://mega.nz/folder/tf5y2ShA#zhCzmRxK9d7d1gzdUIFQZQ), or build from here: https://github.com/SmoothWAN/SmoothWAN/tree/glinet_imagebuilder

leaberry, 2023-09-28:
Well that test image sort of works. There is now the exfat kernel module, but the web interface doesn't work anymore. I flashed the WebUI.tar version from the web interface. When I go to http://172.17.17.2 it redirects me quickly to http://172.17.17.2/cgi-bin/luci/ and then says "Site refused to connect"

I can still ssh in, and I tried 'firstboot' to reset it, but still no luci webui. I suppose I should have told you I'm on a Slate AXT1800 router.

I copied over exfat.ko from /lib/modules though, then sysupgrade'd back to the 0.99.9HF5 release, and copied exfat.ko into /lib/modules/4.4.60 That is preserved during reboot and nicely solves the problem. Now I just need to figure out why samba4-server can't install since libatomic1 is missing from the repos.... But hey, that's a different problem!

I'm patient, this will work until the 1.0 release which I hope will have exfat as a kernel module. Appreciate the help

leaberry, 2023-10-09:
Hmm, you're correct. It modprobes fine, but stacktraces when I plug in an exfat drive. Oh well, I'm happy enough with the speedify bonding that I can't really complain. But next release please add the exfat kernel module (and the sierra wireless one too, then my LTE card will work too!)