SmoothWAN/SmoothWAN #172: Client and master wireless interfaces need to be on the same channel

spacesuitdiver, 2023-12-14:
Master interface needs to be explicitly configured Wi-Fi to the same channel as the AP the client is trying to connect to otherwise it is not reachable.

This seems like it could be a problem when trying to connect to networks with the client interface that have varying channels.

I'm not sure if this is a bug or a limitation but I'm curious your thoughts as to a work around.

Thanks!!

spacesuitdiver, 2023-12-15:
Honestly, I don't understand how to use both client and master at the same time with these routers without just losing the connection and having to reset the whole router (I can't find my ethernet dong lol)

TalalMash, 2023-12-20:
Hi, the AP should follow the client frequency. It's usually not set in the configuration, does `auto` work? Bear in mind that it make take a minute or two due to DFS scan.

spacesuitdiver, 2023-12-20:
I’ll try again with auto… any logging I can help with?

TalalMash, 2023-12-20:
To download the system log, head to Speedify top bar -> Generate Logs.
Extract the archive, it should be `tmp/logreadfile`.