SmoothWAN/SmoothWAN #179: Supporting 4G/5G GL.inet routers

onnimonni, 2024-01-06:
Hey,

This looks very promising project and adds the missing bonding support to the gl.inet routers.

I saw that the current supported devices didn't contain any 4G/5G routers.

I'm willing to contribute how to do this on my own but how difficult would it to be to add support for eg GL-X3000?
https://www.gl-inet.com/products/gl-x3000/

Or if you have other recommendations on how to add 4G/5G modems to the mix I would be very interested to hear more about that and even add to the docs site myself 🙇

TalalMash, 2024-01-06:
Hello!

Unfortunately, GL.iNet removed the source code for building OpenWrt, while the imagebuilder repository exists which supports both Slate AX & Flint it seems to be abandoned when it comes to newer hardware.
Maybe a request to GL.iNet forum to update the image builder would be a start.
Most dongles and ethernet based hotspots are supported. I recommend the wingle series from Huawei for ethernet over USB mode.

Installing Speedify on the existing GL.iNet firmware is possible, since there are many requests regarding this would making an installable package be ok? This isn't related to SmoothWAN however. Official Speedify OpenWrt support seems to be on the way according to the livestreams, but it is delayed.

TalalMash, 2024-01-07:
Check this out, haven't tested it enough yet, feel free to share results:
https://github.com/TalalMash/Unofficial-Speedify-Installer-For-OpenWrt

onnimonni, 2024-01-07:
> Check this out, haven't tested it enough yet, feel free to share results: https://github.com/TalalMash/Unofficial-Speedify-Installer-For-OpenWrt

This looks fantastic. Thanks for sharing 💪 I will order a GL-X3000 and test this out.

SmoothWan would be definitely preferred and looks much more simpler overall to use than the openwrt admin you showed. I'll reach out to GL.iNet about your suggestions and let's see if something will change.

If you do learn more about the official Speedify openwrt support I would be fascinated to hear about it.