SmoothWAN/SmoothWAN #180: New Build Possible?

MXNPD, 2024-01-16:
Is it at all possible to get a new image compiled with all the recent commits? Seems like the last official release was in April of 2023 and some new changes have been committed since then, I'd love to test out the new changes as I typically like running bleeding edge software, even if the new release is just a snapshot and not an official version.

Dogway, 2024-02-28:
Do the latest commits add support for Pi 5?

cr08, 2024-04-02:
I'm also interested in this as well. I've been working with a couple of M.2/USB connected LTE modems in my rig and from what I can see there were a couple of useful MBIM related patches merged into OpenWRT right after the last SmoothWAN release to where MBIM/ModemManager doesn't seem to be a working option in SmoothWAN (have to use QMI) but newer OpenWRT builds it works. It may or may not make much difference in the grand scheme of things but MBIM seems to be the 'standard' nowadays as the required protocol for Windows.

I'm making absolutely ZERO promises/ETAs (this is just a hobby and my day job takes precedence/energy) but I'm gonna see if I can do my own custom build and see where that takes me.

cr08, 2024-04-03:
I just wanted to add a comment and some troubleshooting involved in figuring this out was covered in #183 

I've only done small bits of testing so far but the few key areas including Smoothify itself and interface management are working. YMMV of course. **I highly recommend doing your own testing if this remotely touches anything production level.**

But here's the process I figured how to do your own builds that also work with newer upstream OpenWRT versions.

1.  Follow OpenWRT's instructions for downloading and setting up the image builder for your chosen architecture (My work has been for x86-64 only, but should work similarly for others that SmoothWAN supports): https://openwrt.org/docs/guide-user/additional-software/imagebuilder
2. Clone or download this repo.
3. `packages` should go into the image builder root in the same place as the existing `packages` folder. 
4. Under `devconfigs/x86_64` place the `files` folder, `.config`, and `build.sh` in the image builder root. `.config` should be overwritten.
5. The `build.sh` file needs to be modified as there are a number of packages it wants to install into the image that no longer exist in newer OpenWRT versions for various reasons, some have their functionality replaced elsewhere. There was also a conflict with some wolfssl libraries. I've made a modified build.sh that drops in for x86_64 images: https://github.com/cr08/SmoothWAN/blob/main/devconfigs/x86_64/build.sh
6. From there just set `build.sh` as executable and run it. The rest is hands off and once it has finished you should have a set of images under `./bin/targets/....`

Just want to add a final disclaimer to do your own testing and I take no responsibility if important things break. I'm not sure if there's other important modifications needed to build working images for this, but this process lined up and in my limited testing it boots fine, looks and behaves normal as best as I can tell. This was also tested on the latest stable 23.05.3 build of OpenWRT.

For other architectures, obviously you'd have to modify this procedure to match. Only what's supported by SmoothWAN/is listed under `devconfigs` here should be doable. For anything other than x86-64, you should try building with the stock `build.sh` from this repo. Likely it will error out on some packages it can't install because they are missing from upstream OpenWRT in newer versions. But they may be different from what I saw with x86-64. You'll need to do your own legwork to see what's what. For my run through, there were some translation packages I did not need anyways and some were confirmed to be deprecated/replaced with other packages/functionality. They were simply dropped from the packages list in the `make` line of the `build.sh` script.

TalalMash, 2024-04-10:
There is currently no way to build GL.iNet firmware as the repo went private, and other devices are released together.
Next version will have a lot of refactoring and will use an installable package for GL.iNet devices.
Normal OpenWrt images will remain for Pi/PC but with the included new package and automatic snapshot OpenWrt imagebuilder on Github Actions.