SmoothWAN/SmoothWAN #182: WAN bypass (?) causing interface to drop from Speedify

bt61, 2024-02-03:
This started occurring with a fresh install of Smoothwan a few weeks ago, on a Rpi4 1gb using the same usb-ethernet adapters and cables as prior config, and in troubleshooting I have tried swapping cables and adapters with no change.

I setup exactly according to https://smoothwan.com/wanbyp/ bypass for some clients to my primary wan connection.  When I have bypass set in the _Interfaces>Override ipv4 routing table_, that interface will drop out of Speedify usually in 10 minutes to 1 hour.  If I change Speedify to a different server, that interface drops out every time upon attempting to connect to a new server.  When I turn the _'Override ipv4 routing table'_ to the default _'unspecified'_ the interface never drops, proven up to 48 hours of uptime.

When the interface drops from Speedify, it continues to work and deliver traffic to the clients using the bypass.  Restarting the interface or router does not bring it back into Speedify. I have found that if I can bring it back into Speedify by changing that interface to _ipv4 routing table; unspecified_ and change the (static) IP of the interface.  Then I can go back to Interface and turn the bypass back on, and it will work again for some short amount of time before failing again.

TalalMash, 2024-02-06:
Hi, after the drop post the output of `ip rule show`, `ip route` and `ip route show table BYP#` replacing # with the table that is set for that WAN.
Make sure VPN bypass app is disabled.

bt61, 2024-02-06:
Thank you for the reply!  I did have Bypass active within the Speedify app,  I turned that off and still watched a dropout happen a few minutes later;

```
ip rule show
0:      from all lookup local
1:      from all to 17.0.0.0/8 oif br-lan lookup BYP10
2:      from 172.17.17.71 iif br-lan lookup BYP10
3:      from 172.17.17.72 iif br-lan lookup BYP10
4:      from 172.17.17.73 iif br-lan lookup BYP10
5:      from 172.17.17.74 iif br-lan lookup BYP10
6:      from 172.17.17.50 iif br-lan lookup BYP10
7:      from 172.17.17.52 iif br-lan lookup BYP10
8:      from 172.17.17.53 iif br-lan lookup BYP10
9:      from 172.17.17.80 iif br-lan lookup BYP10
10:     from 172.17.17.81 iif br-lan lookup BYP10
11:     from 172.17.17.82 iif br-lan lookup BYP10
12:     from 172.17.17.76 iif br-lan lookup BYP10
13:     from 172.17.17.77 iif br-lan lookup BYP10
5210:   from all fwmark 0x80000/0xff0000 lookup main
5230:   from all fwmark 0x80000/0xff0000 lookup default
5250:   from all fwmark 0x80000/0xff0000 unreachable
5270:   from all lookup 52
10000:  from 192.168.1.119 lookup BYP10
20000:  from all to 192.168.1.119/24 lookup BYP10
32766:  from all lookup main
32767:  from all lookup default
90012:  from all iif lo lookup BYP10

ip route
0.0.0.0/1 dev connectify0 scope link 
default via 10.202.0.1 dev connectify0 proto static 
default via 192.168.2.1 dev USB_1e2f proto static src 192.168.2.4 metric 75 
10.202.0.0/24 dev connectify0 proto kernel scope link src 10.202.0.2 
123.123.123.123 via 192.168.2.1 dev USB_1e2f 
128.0.0.0/1 dev connectify0 scope link 
172.17.17.0/24 dev br-lan proto kernel scope link src 172.17.17.2 
192.168.2.0/24 dev USB_1e2f proto static scope link metric 75

ip route show table BYP10
default via 192.168.1.1 dev USB_99ad proto static metric 50 
192.168.1.0/24 dev USB_99ad proto static scope link metric 50
```

TalalMash, 2024-02-06:
`1:      from all to 17.0.0.0/8 oif br-lan lookup BYP10` shouldn't exist, is that added intentionally?
`5210:   from all fwmark 0x80000/0xff0000 lookup main` looks like Tailscale is altering the rules.
`192.168.1.0/24 dev USB_99ad proto static scope link` should exist in the default ip route table as well (as in `ip route`)
Uninstall Tailscale, restart and check again to make sure.

bt61, 2024-02-15:
> `1: from all to 17.0.0.0/8 oif br-lan lookup BYP10` shouldn't exist, is that added intentionally? 

Yes, I added that to make Apple updates all bypass Speedify and only go through primary WAN

> `5210: from all fwmark 0x80000/0xff0000 lookup main` looks like Tailscale is altering the rules. `192.168.1.0/24 dev USB_99ad proto static scope link` should exist in the default ip route table as well (as in `ip route`), this is probably the reason why Speedify can't see the interface. Uninstall Tailscale, restart and check again to make sure.

I'll try uninstalling Tailscale, I already have it on my NAS anyway so not really a problem to leave it off my Smoothwan install.

Thanks!

bt61, 2024-02-15:
I uninstalled Tailscale and the problem appears to remain.  

Here is the routes after a restart and not using bypass rules:
```
root@SmoothWAN:~# ip rule show
0:      from all lookup local
1:      from all to 17.0.0.0/8 oif br-lan lookup BYP10
2:      from 172.17.17.71 iif br-lan lookup BYP10
3:      from 172.17.17.72 iif br-lan lookup BYP10
4:      from 172.17.17.73 iif br-lan lookup BYP10
5:      from 172.17.17.74 iif br-lan lookup BYP10
6:      from 172.17.17.50 iif br-lan lookup BYP10
7:      from 172.17.17.52 iif br-lan lookup BYP10
8:      from 172.17.17.53 iif br-lan lookup BYP10
9:      from 172.17.17.80 iif br-lan lookup BYP10
10:     from 172.17.17.81 iif br-lan lookup BYP10
11:     from 172.17.17.82 iif br-lan lookup BYP10
12:     from 172.17.17.76 iif br-lan lookup BYP10
13:     from 172.17.17.77 iif br-lan lookup BYP10
32766:  from all lookup main
32767:  from all lookup default
90012:  from all iif lo lookup BYP10
root@SmoothWAN:~# ip route
0.0.0.0/1 dev connectify0 scope link 
default via 192.168.1.1 dev USB_99ad proto static metric 42 
default via 192.168.2.1 dev USB_1e2f proto static src 192.168.2.4 metric 75 
10.202.0.0/24 dev connectify0 proto kernel scope link src 10.202.0.2 
123.123.123.123 via 192.168.2.1 dev USB_1e2f 
128.0.0.0/1 dev connectify0 scope link 
172.17.17.0/24 dev br-lan proto kernel scope link src 172.17.17.2 
192.168.1.0/24 dev USB_99ad proto static scope link metric 42 
192.168.2.0/24 dev USB_1e2f proto static scope link metric 75 
root@SmoothWAN:~# ip route show table BYP10
root@SmoothWAN:~# root@SmoothWAN:~# ip rule show
-ash: root@SmoothWAN:~#: not found
root@SmoothWAN:~# 0:      from all lookup local
-ash: 0:: not found
root@SmoothWAN:~# 1:      from all to 17.0.0.0/8 oif br-lan lookup BYP10
-ash: 1:: not found
root@SmoothWAN:~# 2:      from 172.17.17.71 iif br-lan lookup BYP10
-ash: 2:: not found
root@SmoothWAN:~# 3:      from 172.17.17.72 iif br-lan lookup BYP10
-ash: 3:: not found
root@SmoothWAN:~# 4:      from 172.17.17.73 iif br-lan lookup BYP10
-ash: 4:: not found
root@SmoothWAN:~# 5:      from 172.17.17.74 iif br-lan lookup BYP10
-ash: 5:: not found
root@SmoothWAN:~# 6:      from 172.17.17.50 iif br-lan lookup BYP10
-ash: 6:: not found
root@SmoothWAN:~# 7:      from 172.17.17.52 iif br-lan lookup BYP10
-ash: 7:: not found
root@SmoothWAN:~# 8:      from 172.17.17.53 iif br-lan lookup BYP10
-ash: 8:: not found
root@SmoothWAN:~# 9:      from 172.17.17.80 iif br-lan lookup BYP10
-ash: 9:: not found
root@SmoothWAN:~# 10:     from 172.17.17.81 iif br-lan lookup BYP10
-ash: 10:: not found
root@SmoothWAN:~# 11:     from 172.17.17.82 iif br-lan lookup BYP10
-ash: 11:: not found
root@SmoothWAN:~# 12:     from 172.17.17.76 iif br-lan lookup BYP10
-ash: 12:: not found
root@SmoothWAN:~# 13:     from 172.17.17.77 iif br-lan lookup BYP10
-ash: 13:: not found
root@SmoothWAN:~# 32766:  from all lookup main
-ash: 32766:: not found
root@SmoothWAN:~# 32767:  from all lookup default
-ash: 32767:: not found
root@SmoothWAN:~# 90012:  from all iif lo lookup BYP10
-ash: 90012:: not found
root@SmoothWAN:~# root@SmoothWAN:~# ip route
[... 96 lines cut ...]
Before:
```
root@SmoothWAN:~# ip route
0.0.0.0/1 dev connectify0 scope link 
**default via 192.168.1.1 dev USB_99ad proto static metric 42** 
default via 192.168.2.1 dev USB_1e2f proto static src 192.168.2.4 metric 75 
10.202.0.0/24 dev connectify0 proto kernel scope link src 10.202.0.2 
**123.123.123.123 via 192.168.2.1 dev USB_1e2f** 
128.0.0.0/1 dev connectify0 scope link 
172.17.17.0/24 dev br-lan proto kernel scope link src 172.17.17.2 
**192.168.1.0/24 dev USB_99ad proto static scope link metric 42** 
192.168.2.0/24 dev USB_1e2f proto static scope link metric 75 
```
After:
```
root@SmoothWAN:~# ip route
0.0.0.0/1 dev connectify0 scope link 
**default via 10.202.0.1 dev connectify0 proto static** 
default via 192.168.2.1 dev USB_1e2f proto static src 192.168.2.4 metric 75 
10.202.0.0/24 dev connectify0 proto kernel scope link src 10.202.0.2 
**123.123.123.123 via 10.202.0.1 dev connectify0** 
128.0.0.0/1 dev connectify0 scope link 
172.17.17.0/24 dev br-lan proto kernel scope link src 172.17.17.2 
192.168.2.0/24 dev USB_1e2f proto static scope link metric 75 
```

JustKuz, 2024-02-15:
I too see this issue. I have a vlan I have attempted to bypass to a specific WAN. Within a few minutes of adding the BYP * override to the WAN it drops off from speedify.