SmoothWAN/SmoothWAN #184: Intel N100 mini PC, no network ports found

Orbit77, 2024-08-14:
Attempting to install SmoothWan on an Intel N100 based device, no WiFi, 4 Ethernet ports.  The plan is to bond 3 slow LTE connections and use the 4th port to share the bonded connection with the rest of the network.  Device is similar to this (search: 4x i226-V 2.5G LAN 12th Generation Intel N100 Firewall Router N5105)

Upon booting from the image, after the 2 expected reboots, the device has no IP and no connection.  Logging on to the device, none of the 4 ports are detected, it seems like missing drivers from the image.  Has anyone managed to get SmoothWan working on a similar device?

sdacheson78, 2024-08-14:
I would use proxmox and run as a virtual machine.  It makes running things
easier for smoothwan because it has the the drivers for virtual NIC's.

you can also see things through the web interface easier.

Just a thought.

Orbit77, 2024-08-15:
Thanks, that’s a good idea.

I had read about performance issues with Proxmox, but I doubt my
connections are quick enough to stress the machine, even virtualised.

Orbit77, 2024-08-21:
> I would use proxmox and run as a virtual machine. It makes running things easier for smoothwan because it has the the drivers for virtual NIC's. you can also see things through the web interface easier. Just a thought.
> […](#)

To confirm, this worked great.  The documentation for setting up using Proxmox is excellent and I'm getting consistently over 100 Mbps bonding 3 cellular connections and unlike OpenMPTCPRouter, the Netflix bypass means streaming services continue to work.  The streaming bypass and excellent docs make SmoothWan the best solution for me.  Thanks.

closed 2024-08-21 by Orbit77